pkerspe/ESP-StepperMotor-Server #5: Reboot after successful run

anttitai, 2020-07-05:
After run  I'll get following information : 

Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0xfefefefe  PS      : 0x00060b30  A0      : 0x800d944a  A1      : 0x3ffdb810  
A2      : 0xfefefefe  A3      : 0x00000001  A4      : 0x00001388  A5      : 0x00001388
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x8014d620  A9      : 0x3ffdb7f0  
A10     : 0x02000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001
A14     : 0x00000000  A15     : 0x3ffd91e0  SAR     : 0x00000007  EXCCAUSE: 0x00000014  
EXCVADDR: 0xfefefefc  LBEG    : 0x400029ac  LEND    : 0x400029cb  LCOUNT  : 0x00000000  

Backtrace: 0x7efefefe:0x3ffdb810 0x400d9447:0x3ffdb830 0x40088e99:0x3ffdb850

**Decoded results:**
Decoding stack results
0x400d9447: ESPStepperMotorServer_MotionController::processMotionUpdates(void*) at \ESP-StepperMotor-Server\src\ESPStepperMotorServer_MotionController.cpp line 76
0x40088e99: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

Rebooting...

pkerspe, 2020-07-05:
Hello @anttitai
A assume your compiling in Arduino IDE. Since it does not manage dependencies properly you might probably be using the master branch version of the ESP-FlexyStepper library.
The issue you describe is probably only occurring when you start moving the stepper motor, right?
If so, you should install ESP-FlexyStepper version 1.2 and not the last version from master branch.
The master branch is currently in a state that has issues with new features.
Please provide more details if the issue persists after installing ESP-FLexyStepper 1.2

In general I recommend using a proper IDE that can manager dependencies a bit better than Arduino IDE, I suggest taking a look into PlatformIO. Takes only minutes to install and get ready to go and has way more features than Arduino IDE. But that is just a side note, once you update the Library version of ESP-FlexyStepper you should be able to run it also using the Arduino compiled code

pkerspe, 2020-07-05:
closed since it has been confirmed working by @anttitai in the duplicate of this issue

closed 2020-07-05 by pkerspe

lazyzero, 2020-07-07:
The problem seams to be still present. As there is no version 1.2 available of the ESP-FLexyStepper I tried with the 1.3
Any recommondations?

Best Christian

> [INFO] Motion Controller task started
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0xfefefefe  PS      : 0x00060f30  A0      : 0x800d9432  A1      : 0x3ffdba10
A2      : 0xfefefefe  A3      : 0x00000001  A4      : 0x000001f4  A5      : 0x000001f4
A6      : 0x00000003  A7      : 0x00060223  A8      : 0x8014d36c  A9      : 0x3ffdb9f0
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001
A14     : 0x00000000  A15     : 0x3ffd93e0  SAR     : 0x00000020  EXCCAUSE: 0x00000014
EXCVADDR: 0xfefefefc  LBEG    : 0x400029ac  LEND    : 0x400029cb  LCOUNT  : 0x00000000

Backtrace: 0x7efefefe:0x3ffdba10 0x400d942f:0x3ffdba30 0x40088e99:0x3ffdba50

Rebooting...

pkerspe, 2020-07-07:
please post your script with the main function / Setup function and which exact version of ESP-StepperMotor-Server you have installed.
Also please provide information about which hardware you are using (revision and memory).
The decoded backtrace would also help a lot to identify the exact position in the code.

pkerspe, 2020-07-08:
@lazyzero FYI I will work this weekend on a new release of both, ESP-FlexyStepper and ESP-StepperMotor-Server, to ease dependency management issues

lazyzero, 2020-07-08:
> please post your script with the main function / Setup function and which exact version of ESP-StepperMotor-Server you have installed.
> Also please provide information about which hardware you are using (revision and memory).
> The decoded backtrace would also help a lot to identify the exact position in the code.
> 
> ESP-FlexyStepper 1.2 is available here: https://github.com/pkerspe/ESP-FlexyStepper/releases/tag/v1.2

Hi @pkerspe ,

I used the code from the readme. 
`void setup()
{
  Serial.begin(115200);
  stepperMotorServer = new ESPStepperMotorServer(ESPServerRestApiEnabled | ESPServerWebserverEnabled | ESPServerSerialEnabled);
  stepperMotorServer->setWifiCredentials(wifiName, wifiSecret);
  stepperMotorServer->setWifiMode(ESPServerWifiModeClient); //start the server as a wifi client (DHCP client of an existing wifi network)
  stepperMotorServer->start();
}

void loop()
{
}`

This are the importand infos from platformio:

`
PLATFORM: Espressif 32 1.12.2 > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
...
Dependency Graph
|-- <ESP-StepperMotor-Server> 0.2.4
|   |-- <ESP Async WebServer> 1.2.3
|   |   |-- <AsyncTCP> 1.1.1
|   |   |-- <FS> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <ArduinoJson> 6.15.2
|   |-- <ESP-FlexyStepper> 1.3
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <ArduinoJson> 6.15.2
|   |-- <FS> 1.0
|   |-- <HTTPClient> 1.2
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|-- <ESP-FlexyStepper> 1.3`

I use the platfomio registry for fetching libraries. and If you look on https://platformio.org/lib/show/7427/ESP-FlexyStepper you will see that the version 1.2 is not listed there. It is 1.0, 1.1 and 1.3

I will wait for the weekend then ;-)

Best Christian

pkerspe, 2020-07-12:
Hi Christan (@lazyzero),

I released a new version of ESP-FlexyStepper that fixes the kernel panic: https://github.com/pkerspe/ESP-FlexyStepper/releases/tag/v1.3.1

You can install it manually by downloading the zip file or you wait 1-2 days until it is indexed in the Platform IO Library manager (as soon as it shows up here, you can also install it via the platformIO UI: https://platformio.org/lib/show/7427/ESP-FlexyStepper)