pkerspe/ESP-StepperMotor-Server #6: esp32 reboot on motor motion

taloot, 2020-07-11:
it gives me core 0 panic  in esp32 terminal

its wemos

taloot, 2020-07-11:
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
22:47:39.307 -> Core 0 register dump:
22:47:39.307 -> PC      : 0xfefefefe  PS      : 0x00060130  A0      : 0x800d9a77  A1      : 0x3ffdb700  
22:47:39.307 -> A2      : 0xfefefefe  A3      : 0x00000001  A4      : 0x000001c1  A5      : 0x000001c1  
22:47:39.307 -> A6      : 0x00000003  A7      : 0x00060023  A8      : 0x800ee264  A9      : 0x3ffdb6e0  
22:47:39.307 -> A10     : 0x00000004  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001  
22:47:39.307 -> A14     : 0x00000000  A15     : 0x3ffd90c0  SAR     : 0x0000001e  EXCCAUSE: 0x00000014  
22:47:39.354 -> EXCVADDR: 0xfefefefc  LBEG    : 0x400029ac  LEND    : 0x400029cb  LCOUNT  : 0x00000000  
22:47:39.354 -> 
22:47:39.354 -> Backtrace: 0x7efefefe:0x3ffdb700 0x400d9a74:0x3ffdb720 0x40088e99:0x3ffdb740
22:47:39.354 -> 
22:47:39.354 -> Rebooting...

pkerspe, 2020-07-11:
Please use ESP-FlexyStepper in version 1.2 (manually installed)

Current 1.3x causes this issue.
Fix will be provided in the next days with version 1.4 of ESP-FlexyStepper

taloot, 2020-07-11:
any how can install it... or whats the package number for... as i cant install it normal why,,, and thanks for ur quick reply

pkerspe, 2020-07-12:
I released a new version of ESP-FlexyStepper that fixes the kernel panic: https://github.com/pkerspe/ESP-FlexyStepper/releases/tag/v1.3.1

You can install it manually by downloading the zip file or you wait 1-2 days until it is indexed in the Platform IO Library manager (as soon as it shows up here, you can also install it via the platformIO UI: https://platformio.org/lib/show/7427/ESP-FlexyStepper)

closed 2020-07-12 by pkerspe

taloot, 2020-07-12:
> I released a new version of ESP-FlexyStepper that fixes the kernel panic: https://github.com/pkerspe/ESP-FlexyStepper/releases/tag/v1.3.1
>
> You can install it manually by downloading the zip file or [...]

thanks i install 1.2 and it worked
thanks for ur gr8 time

i use it to control the CNC fiber laser auto foucas head,,