pkerspe/ESP-StepperMotor-Server #7: Add Option for a mechanical brake

BenSpex, 2020-07-31:
**Is your feature request related to a problem? Please describe.**
I would like to drive a motor with an integrated brake. The brake control is connected to the ESP32 via a Transistor. 
PIN HIGH --> Brake is disabled, PIN LOW Brake is enabled. 

**Describe the solution you'd like**
Option to configure a Brake-pin with a time out. 
If the motor has not moved for xx minutes the brake is enabled, PIN LOW, and the motor disabled. 
This would save energy and increase safety. 

**Describe alternatives you've considered**
I'm currently running a very basic custom script to control the motor. But this offers much more functionality and is so much either to use and configure

pkerspe, 2020-07-31:
Thanks for this well defined feature request.
Just one question:
You wrote, that the break should be enabled after the motor stood still for x minutes, is that correct or do you mean the break should be disabled after the motor stood still for some minutes?

BenSpex, 2020-08-01:
Thanks for the quick reply. 

Some stepper motors include a break for example [https://en.nanotec.com/products/616-as5918l4204-enm24b/](url) 
The brake is typically required if the stepper motor is moving something that is vertically mounted. As long as the motor is enabled and current is provided the holding torque of the motor will keep everything in place. 
Once the power is turned off, or in case of a power failure the brake kicks in to hold the load in place. 

At the same time power consumption is much lower with enabled brake and disabled motor. 

So you are correct you want to enhance the brake as soon as the motor comes to a stop. But to avoid unnecessary stress to on the mechanics a time out would be preferred. 

Normal motor operation. Motor Enabled, brake disabled. 

IDLE, Brake enabled, Motor Disabled (saving power) 

Emergency (Enable brake)

pkerspe, 2020-08-01:
now that's a fancy-schmancy stepper motor :-)
Ok I will add a configuration option to define a PIN number for the break, define active high or active low for this pin and a delay to engange the break after motion is supposed to stop and a delay after which the break is disabled if the motor hasn't moved for a defined period. Both delays shall be optional (default is no delay for engaging the break, and never disable break as long as the motor is stopped).

Also engage break in case of emergency stop and release again once new target position is set.

All correct?

pkerspe, 2020-08-01:
I updated the ESP-FlexyStepper Library to support the brake pin and release version 1.4 (https://github.com/pkerspe/ESP-FlexyStepper/releases/tag/v1.4)
Before proceeding with updating the ESP-StepperMotor-Server I have to wait for the ESP-FlexyStepper release to be updated in the PlatformIO repo, otherwise people will get errors when trying to compile the ESP-StepperMotor-Server since the dependency is missing in the required version. Will keep you updated.

The UI has already been adopted and released in version 0.3: https://github.com/pkerspe/ESP-StepperMotor-Server-UI/releases/tag/v0.3

pkerspe, 2020-08-03:
@BenSpex I just release version 0.3.1 of the Stepper Server, this includes an updated UI in the Example folder and the new functions in the Backend to configure a brake PIN as well as engage and release delays in milliseconds.
I did not perform real tests on this new function, more of theoretical debugging. 
please let me know if the functionality is as expected and all works as intended

pkerspe, 2020-08-03:
@BenSpex you might want to go straight to 0.3.2 since it also features some improvement of the UI view "Control" in regard to near-time position updates.
I now close this feature request, since it should be completed. Please let me know if there are any issues

closed 2020-08-03 by pkerspe

BenSpex, 2021-08-13:
Hey @pkerspe Sorry I never thanked you for this feature :) It did work

pkerspe, 2021-08-14:
glad to hear, thanks for the feedback!