pkerspe/ESP-StepperMotor-Server #8: switches to trigger movement

savehanges, 2020-08-08:
There is a future feature called "switches to trigger movement macros". is this available for beta testing ?
i need to run StepperMotor-Server for custom movements trigger by sensors.

pkerspe, 2020-08-08:
sorry, not yet implemented. But maybe you can describe your use case a bit further?
It is actually quite simple to write your own implementation by 
1.) registering an interrupt handler for the switch you want to use to trigger a movement
2.) in the ISR you simply get the stepper motor configuration of the stepper you want to move and call the setTargetPositionInSteps function on the flexyStepper Instance. e.g 
````
ESP_FlexyStepper *stepper = stepperMotorServer->getCurrentServerConfiguration()->getConfiguredFlexySteppers()[0];
stepper->setTargetPositionInSteps(1000);
````

savehanges, 2020-08-09:
Thanks for quick reply. basically i want to move 3 axis. 1st one moves until it find the sensor and then move back. 2nd one move after the fist axis touch the sensor. 3rd axis just moving between limit switches with a trigger sensor regardless of other axis.

closed 2020-08-30 by savehanges