pkerspe/ESP-StepperMotor-Server #9: Cannot save configiguration

videobuff, 2020-08-31:
**Describe the bug**
The configuration is not saved to SPIFFS. 

**Environment**
Platformio
Expressif 1.12.14
ArduinoJson@6.16.1

**To Reproduce**
See the YouTube movie. I can do everything, but it wont save, and there is no proper id given, all are 0

**Expected behavior**
Save it to SPIFFS to be able to recall later

**Screenshots**
See the movie at youtube
[https://youtu.be/7pCqvhruycg](https://youtu.be/7pCqvhruycg)

[image]

[image]

[image]
[image]

videobuff, 2020-08-31:
Also it does not work from the CLI. If i recall the config, it is always empty.
[image]

pkerspe, 2020-08-31:
Thank you for the detailled description. Unfortunaltey the video quality is to low to recognize anything that is stated in the console/Serial output.
Yet I can see different issues that seem to point to some incomplete UI in the SPIFFS.
No Icons are showing for Steppers, Switches and Encoders also your output "listing files in root folder" is not complete.
It shows only 5 files, yet in the current version of the UI it should be 10.
Can you please post the output of your call to "Upload Filesystem Image" in platformio?

Here the output I get when uploading the current version of the UI:
```CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.12.4 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP-StepperMotor-Server> 0.4.0
|   |-- <ESP Async WebServer> 1.2.3
|   |   |-- <AsyncTCP> 1.1.1
|   |   |-- <FS> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <ArduinoJson> 6.16.1
|   |-- <ESP-FlexyStepper> 1.4.3
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <ArduinoJson> 6.16.1
|   |-- <FS> 1.0
|   |-- <HTTPClient> 1.2
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
Building in release mode
Building SPIFFS image from '/Users/paul.kerspe/Documents/PlatformIO/Projects/ESP-StepperMotor-Server-UI/data' directory to .pio/build/esp32dev/spiffs.bin
/favicon.ico
/index.html
/js/app.js.gz
/config.json
/img/rotaryEncoderWheel.svg
/img/stepper.svg
/img/emergencyStopSwitch.svg
/img/switch.svg
/img/logo.svg
/upload.html.gz
Looking for upload port...
Use manually specified: /dev/cu.SLAB_USBtoUART
Uploading .pio/build/esp32dev/spiffs.bin
esptool.py v2.6
Serial port /dev/cu.SLAB_USBtoUART
Connecting.....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, Coding Scheme None
MAC: 30:ae:a4:8b:bb:e8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1507328 bytes to 255642...

Writing at 0x00290000... (6 %)
Writing at 0x00294000... (12 %)
Writing at 0x00298000... (18 %)
Writing at 0x0029c000... (25 %)
Writing at 0x002a0000... (31 %)
Writing at 0x002a4000... (37 %)
Writing at 0x002a8000... (43 %)
Writing at 0x002ac000... (50 %)
Writing at 0x002b0000... (56 %)
Writing at 0x002b4000... (62 %)
Writing at 0x002b8000... (68 %)
Writing at 0x002bc000... (75 %)
Writing at 0x002c0000... (81 %)
Writing at 0x002c4000... (87 %)
Writing at 0x002c8000... (93 %)
Writing at 0x002cc000... (100 %)
Wrote 1507328 bytes (255642 compressed) at 0x00290000 in 6.2 seconds (effective 1959.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...````

pkerspe, 2020-08-31:
Also I suggest to turn on debugging in your program at the point where you create the Stepper Motor instance.
If you used something like this:
```    stepperMotorServer = new ESPStepperMotorServer(ESPServerRestApiEnabled | ESPServerWebserverEnabled | ESPServerSerialEnabled, ESPServerLogLevel_INFO);````
Then change it to this:
```    stepperMotorServer = new ESPStepperMotorServer(ESPServerRestApiEnabled | ESPServerWebserverEnabled | ESPServerSerialEnabled, ESPServerLogLevel_DEBUG);````

OR you can use the serial console of the ESP Stepper Motor server and just type:
```ll=4``to set the log level to all.
Then try again in the UI to create a new config (just add a stepper with dummy data, no need to create a full fledged config before clicking save).

If this does not work, and you only need a fixed configuration, you can also create it manually in the "config.json" that you uploaded along with the UI.

A simple sample configuration would look like this:
````{
    "serverConfiguration": {
        "port": 80,
        "apName": "ESP-StepperMotor-Server",
        "apPassword": "ServerPassword",
        "wififMode": 2,
        "wifiSsid": "",
        "wifiPassword": ""
    },
    "stepperConfigurations": [
        {
            "id": 0,
            "stepPin": 32,
            "directionPin": 33,
            "name": "X-Axis",
            "stepsPerRev": 200,
            "stepsPerMM": 100,
            "microsteppingDivisor": 1,
            "rpmLimit": 1000
        }
    ],
    "switchConfigurations": [
        {
            "id": 0,
            "ioPin": 14,
            "stepperIndex": 0,
            "switchType": 64,
            "name": "Limit 1 X-Axis",
            "switchPosition": -1
        }
    ],
    "rotaryEncoderConfigurations": [
    ]
}````
just copy this into the config.json file and upload to see if reading the configuration works at all.

pkerspe, 2020-08-31:
plase also post your main.cpp (without your wifi credentials ;-) if any)

videobuff, 2020-08-31:
The error was completely on my side. When I moved to my win10 PC, installed platformio and the project according to your instructions, all went well. Thanks very much. I have build an antenna rotor controller using a stepper motor in stead of the old fashioned 2 fase motor. using your library i should be able to control it in all aspects.

Erik

[image]

closed 2020-09-01 by pkerspe