pkerspe/ESP-StepperMotor-Server #10: Robot arm?

JuniorGamingTime, 2020-09-07:
I'm not the best with code so I like the simplicity of this library, but I was curious if this could be used with normal 3d printer drivers? Like the a4988. I'm thinking of building a robot arm with 6 steppers and 2 servos. Would this library be useful for this?

pkerspe, 2020-09-07:
I never worked with the a4988 driver series, but it seems it has a step and a dir pin, so it should work with the library indeed.
Every driver that works with just these two pins will work with the library. 

In the unlikely case the driver requires 5V signals (it seems the a4988 works just fine with 3.3V) you need to integrate a level shifter, but I assume even 5V operated drivers should work with the 3.3v signals provided by the ESP.

pkerspe, 2020-09-07:
Oh, servos are not yet supported in the library, but I was already thinking about that too for a future release.

JuniorGamingTime, 2020-09-08:
oh thats good to know might try some TMC2100s then so its more quite for the smaller stepper, and for something like the nema 23s I can use those bigger drivers. I've been wanting a little challenge with 3D modeling so decided on a robot so please keep us all updated if you add servo support soon. With your library this would be a huge weight lifted and would make the robot arm more of a possibility, thanks alot

JuniorGamingTime, 2020-09-12:
Hi so im finally for the ESP32 and some of the motor controllers to start my robot arm project. im having issues even starting with the install. 

I'm using PlatformIO in visual studios and used the "Github Desktop" App. I cant seem to find the platformio.ini file or the "main.cpp" file. I just made the ini file would i need to do the same for the main.cpp file? and how does the code know where i plug in stuff and how many steppers i have? is it all via the GUI?

Thanks in advance really excited to get this started :D

JuniorGamingTime, 2020-09-13:
So i managed to flash the esp32 but im getting this error and the UI is not working:
[vfs_api.cpp:22] open(): File system is not mounted

pkerspe, 2020-09-13:
It seems you did not initialise the SPIFFS (file system of the ESP).
You need to format the file system or use the upload filessystem function in platformio to upload the contents of the "data" folder.

pkerspe, 2020-09-13:
Oh, and the setup of the pins / connected Stepper drivers can be done via the user interface once you are done with the SPIFFS upload.
You can also configure it via code (hardcoded config), but the UI is probably the preferred way.

JuniorGamingTime, 2020-09-13:
Cool ill give that a try here soon ive never done this with the ESP32 and new to PlatformIO

JuniorGamingTime, 2020-09-13:
So im trying to "Build Filesystem Image" and i get this error
[image]

pkerspe, 2020-09-13:
copy the data folder from the example directory into your project root folder and try again

JuniorGamingTime, 2020-09-13:
added the files and still getting the same error 
[image]

JuniorGamingTime, 2020-09-13:
Ok finally seems like i got the UI to show up ill keep updating here is any issues

pkerspe, 2020-09-13:
You need to copy the folder "data" into the root, not only the files from within the data folder.
But I guess you figured that out by now :-)

JuniorGamingTime, 2020-09-13:
So i have everthing connected (just a nema 23 with a TB6600) like in the diagram but the motor is not turning. is this setup right?
[image]

pkerspe, 2020-09-14:
first of all you should probably not use PINs 1 and 3, check the Pin Out Diagramm of the ESP32 and which Pins are good to use: https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

It is safer to use e.g. Pin 16 and 17 since it does not have any other functions during boot process or for serial communication etc.

If it still does not work, check your power supply for the stepper driver. Also if you had a oscilloscope you could check the output of the configured pins, if signals are being send when you send some move commands to the ESP Stepper Motor Server

JuniorGamingTime, 2020-09-14:
so i tested 16, 17 and 33,32 for the pins and still nothing. i swapped the driver and no change. if i unplug the enable pin the stepper locks up so it seems like both are working. and how can i see the serial console? and power supply is at a solid 12v and the wiring is the same as in the example just different step and dir pins.

pkerspe, 2020-09-14:
Can you post a wiring diagramm? Especially for the enable pin you are talking about. It should be permanently connected to ground or vcc e.g. 3.3 or 5V (depending on the stepper driver).

To connect to the serial console you can use any serial tool you want or just use the serial monitor in PlatformIO. 
See platformIO manual on how to start the serial console and configure the baud rate in platformio.ini

maybe also the 3.3V of the ESP is not enough and you need to integrate a level shifter to 5V for step and dir pin signales.

Test completely without the ESP32:
Do you get a movement when you connect step pin of the driver directly to 3.3V and then tap the dir pin of the driver to the 3.3V power rail repeatedly? The stepper should then perform a single step each time you connect to vcc. If this does not work, your wiring is wrong in general.

Also:
Did you connect the ground of the ESP with the ground of the Stepper driver?

It would be easier if you post a complete wiring diagramm

JuniorGamingTime, 2020-09-14:
Sorry i cant seem to understand or get the part (like the tb6600) into Fritzing.
on the TB6600:
ENA-, DIR-. PUL- are going to negative 5V
ENA+ to +5V
DIR+ to D32 (GPIO32)
PUL+ to D33 (GPIO33)

ESP32:
VIN to +5V
GND to -5V

i tested with ground of the ESP to the Ground of the Stepped driver with no change

Also i get this in the serial. only sends that when i refresh 
[image]

pkerspe, 2020-09-14:
are you sure you have negative 5V? Or do you mean Ground?

Have you tried connecting DIR+ to 5V permanently and then tapping (pulsing) PUL+ to 5V as well multiple times? (nothing connected to the ESP32)
Each time you touch +5V with the PUL+ line, the stepper should perform a single step. If not, then your wiring is wrong (maybe 12V Power supply not connected correct, you did not mention it at all in your wiring, or the stepper is not corrected correctly to the stepper driver)

pkerspe, 2020-09-14:
also I am a bit confused: you wrote at the beginning, that you want to use a4988 driver. Now you wrote you connected a tb6600 driver.
When using such a driver, also make sure you set the right amperage (so you don't fry your stepper motor) and that you dissable microstepping for the initial tests.

pkerspe, 2020-09-14:
I just checked the tb6600 datasheet and you do not NEED to connect the ENA+/ENA- pin at all, so for the sake of simplicity I would avoid this step.

JuniorGamingTime, 2020-09-14:
seems like that was the problem. i got it to spin.
and yeah i just got 4 TB6600 for the Nema 23s and ill be using a4988 for the Nema17 and maybe a Nema11

JuniorGamingTime, 2020-09-14:
Now im not sure about this new issues. 
ive never worked with Nema23 or TB6600s but it seems like the stepper is missing steps and its not smooth. i have the microstepping off and the current the same at the Stepper. could it be a bad stepper or the TB6600?

pkerspe, 2020-09-14:
Lost steps could have multiple reasons:
- speed (pulses per second) is set to high
- amperage for the Stepper motor is to low
- voltage for the stepper motor is to low
- load on the stepper is to high 
And probably some more reasons.

Check the data sheet of your stepper motor for max revs/min, required voltage and ampere count and try first with only one stepper motor configured in the esp stepper motor server.

What speed did you set for the movement? Is there any load on your stepper motor or is it spinning freely?

JuniorGamingTime, 2020-09-14:
The stepper id spinning freely, Revs/min, Volts and Amps are all correct. Im also doing one stepper for now

[image]

[image]

JuniorGamingTime, 2020-09-14:
seems to only happen at slower speeds too

pkerspe, 2020-09-15:
you are trying to move the stepper with 2000 steps per second for a distance of 200 steps and an acceleration of 500 steps per second. So you you will never reach the top speed on this short distance.
Also did you check your motors specs in regards to max revs?
2000 Steps and Full Step mode with 200 steps per rev would result in 10 revs per second or 600 revs per minute. Ususally stepper motors loose a lot of torque at higher speeds (depending on the stepper of course, so you need to check your datasheets) At roughly around 1000-1500 revs/minute the power curve of most stepper motors breaks down.
Plus you are running at only 12V which is pretty much the lower end for most stepper motors. Most data sheets print power curves at 24v or 30v (or even higher for bigger motors).

Try using a 24V power supply if possible.

For smoother / quieter operation you need to enable microstepping, but of course this will limit your max speed when running with multiple stepper drivers since more step signals need to be generated per timeframe.

JuniorGamingTime, 2020-09-15:
I have a small test bench that can do 24v max 5 amps itll be enough for testing one stepper (the arm should have around 6 maybe 7 steppers) plus i only really need full steps do so me using a 3d printer 8:1 planetary gear to get more torque for the lower parts of the arm.

The data sheet seems reallt basic and cant really find the Max revolutions, bit wierd.
https://www.omc-stepperonline.com/nema-23-bipolar-1-8deg-1-26nm-178-4oz-in-2-8a-2-5v-57x57x56mm-4-wires.html

JuniorGamingTime, 2020-09-15:
No change with 24v but i did notice that it seems to be only pulling half the amps around 1.3amps. i even set the TB6600 to the max current just to test and it went up too. 1.5amps i have swapped bettew the 4 TB6600 i have with no change. i should be getting  my second nema 23 tomorrow and then i can test if its the Stepper

JuniorGamingTime, 2020-09-15:
So out of curiosity i tested with the NEMA 17 i have and the same issue happens. All i did was change the current for the smaller nema17 the acceleration to full speed is jerky and when it reaches the max speed set its like does a small jolt on the end of a full revolution and could be the reason of the missed steps and positioning being off

pkerspe, 2020-09-15:
Just to be sure, can you post the full serial output from botting the ESP to issuing the move command? Maybe you could even turn on debugging in your main.cpp file and upload it again to get more details.

JuniorGamingTime, 2020-09-15:
this is what i get

[DEBUG] POST called/api/steppers/moveby with parameters: id=0, value=500, unit=steps, speed=800, accel=500, decel=500
[DEBUG] GET called/api/status

pkerspe, 2020-09-15:
please post the complete output from booting the esp until the point you call the endpoint, not only the the output when calling the endpoint

JuniorGamingTime, 2020-09-15:
im guessing this is what you mean correct?

[INFO] Setting log level to INFO
[INFO] Loading configuration file /config.json from SPIFFS
[INFO] 1 stepper configuration entry loaded from config file
[INFO] 0 switch configuration entries loaded from config file
[INFO] 0 rotary encoder configuration entries loaded from config file
[INFO] Starting ESP-StepperMotor-Server (v. 0.3.3)
[INFO] Trying to connect to WiFi with SSID 'FBIVan' ....
[INFO] Connected to network with IP address 192.168.50.132
[INFO] Listing files in root folder of SPIFFS:
[INFO] File: /favicon.ico (318) -1
[INFO] File: /img/emergencyStopSwitch.svg (11442) -1
[INFO] File: /img/logo.svg (25066) -1
[INFO] File: /img/rotaryEncoderWheel.svg (13750) -1
[INFO] File: /img/stepper.svg (22739) -1
[INFO] File: /img/switch.svg (19709) -1
[INFO] File: /index.html (678) -1
[INFO] File: /js/app.js.gz (210333) -1
[INFO] File: /config.json (346) 1003
[INFO] Starting webserver on port 80
[INFO] Webserver started, you can now open the user interface on http://192.168.50.132:80/
[INFO] Command Line Interface started, registered 22 commands. Type 'help' to get a list of all supported commands
[INFO] Motion Controller task started
[INFO] Setting log level to DEBUG
[DEBUG] Checking SPIFFS for existance and free space
[DEBUG] SPIFFS started
SPIFFS stats:
Total bytes: 1378241
bytes used: 308981
bytes free: 1069260
[INFO] Loading configuration file /config.json from SPIFFS
[DEBUG] File loaded and deserialized
[INFO] 1 stepper configuration entry loaded from config file
[INFO] 0 switch configuration entries loaded from config file
[INFO] 0 rotary encoder configuration entries loaded from config file
{
  "serverConfiguration": {
    "port": 80,
    "wififMode": 2,
    "wifiSsid": "FBIVan",
    "wifiPassword": "*****",
    "apName": "p[\r@",
    "apPassword": "*****"
  },
  "stepperConfigurations": [
    {
      "id": 0,
      "name": "test",
      "stepPin": 33,
      "directionPin": 32,
      "stepsPerRev": 200,
      "stepsPerMM": 1,
      "microsteppingDivisor": 1,
      "rpmLimit": 1200
    }
  ],
  "switchConfigurations": [],
  "rotaryEncoderConfigurations": []
}
[DEBUG] ESPStepperMotorServer_RestAPI instance created
[DEBUG] Motor Controller created
[WARNING] ESPStepperMotorServer must be used as a singleton, do not instanciate more than one server in your project
[DEBUG] GET called/api/status
[DEBUG] POST called/api/steppers/moveby with parameters: id=0, value=500, unit=steps, speed=800, accel=500, decel=500
[DEBUG] GET called/api/status

pkerspe, 2020-09-15:
can you please also post your main.cpp file? The Warning `[WARNING] ESPStepperMotorServer must be used as a singleton, do not instanciate more than one server in your project` is bugging me a bit

JuniorGamingTime, 2020-09-15:
#include <ESPStepperMotorServer.h>

ESPStepperMotorServer *stepperMotorServer;

const char *wifiName = "FBIVan";       // enter the SSID of the wifi network to connect to
const char *wifiSecret = "********"; // enter the password of the the existing wifi network here

void setup()
{
  Serial.begin(115200);
  stepperMotorServer = new ESPStepperMotorServer(ESPServerRestApiEnabled | ESPServerWebserverEnabled | ESPServerSerialEnabled);
  stepperMotorServer->setWifiCredentials(wifiName, wifiSecret);
  stepperMotorServer->setWifiMode(ESPServerWifiModeClient); //start the server as a wifi client (DHCP client of an existing wifi network)
  stepperMotorServer->start();
  stepperMotorServer = new ESPStepperMotorServer(ESPServerRestApiEnabled | ESPServerWebserverEnabled | ESPServerSerialEnabled, ESPServerLogLevel_DEBUG);
}

void loop()
{
}

pkerspe, 2020-09-15:
you must not create to instances of the ESPStepperMotorServer.
Please move the line ````
stepperMotorServer = new ESPStepperMotorServer(ESPServerRestApiEnabled | ESPServerWebserverEnabled | ESPServerSerialEnabled, ESPServerLogLevel_DEBUG);```
up and replace the line ````
stepperMotorServer = new ESPStepperMotorServer(ESPServerRestApiEnabled | ESPServerWebserverEnabled | ESPServerSerialEnabled);``` with this line

JuniorGamingTime, 2020-09-15:
seems to be smoother now, acceleration is a bit noise kind of like a crunch sound 

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on COM9  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
[INFO] Setting log level to DEBUG
[DEBUG] Checking SPIFFS for existance and free space
[DEBUG] SPIFFS started
SPIFFS stats:
Total bytes: 1378241
bytes used: 308981
bytes free: 1069260
[INFO] Loading configuration file /config.json from SPIFFS
[DEBUG] File loaded and deserialized
[INFO] 1 stepper configuration entry loaded from config file
[INFO] 0 switch configuration entries loaded from config file
[INFO] 0 rotary encoder configuration entries loaded from config file
{
  "serverConfiguration": {
    "port": 80,
    "wififMode": 2,
    "wifiSsid": "FBIVan",
    "wifiPassword": "*****",
    "apName": "p[\r@",
    "apPassword": "*****"
  },
  "stepperConfigurations": [
    {
      "id": 0,
      "name": "test",
      "stepPin": 33,
      "directionPin": 32,
      "stepsPerRev": 200,
      "stepsPerMM": 1,
      "microsteppingDivisor": 1,
      "rpmLimit": 1200
    }
  ],
  "switchConfigurations": [],
  "rotaryEncoderConfigurations": []
}
[DEBUG] ESPStepperMotorServer_RestAPI instance created
[DEBUG] Motor Controller created
[INFO] Starting ESP-StepperMotor-Server (v. 0.3.3)
[INFO] Trying to connect to WiFi with SSID 'FBIVan' ....
[INFO] Connected to network with IP address 192.168.50.132
[INFO] ESPStepperMotorServer WiFi details:
[INFO] WiFi status: server acts as wifi client in existing network with DHCP
[INFO] SSID: FBIVan
[INFO] IP address: 192.168.50.132
[INFO] Strength: -29 dBm
[INFO] Listing files in root folder of SPIFFS:
[INFO] File: /favicon.ico (318) -1
[INFO] File: /img/emergencyStopSwitch.svg (11442) -1
[INFO] File: /img/logo.svg (25066) -1
[INFO] File: /img/rotaryEncoderWheel.svg (13750) -1
[INFO] File: /img/stepper.svg (22739) -1
[INFO] File: /img/switch.svg (19709) -1
[INFO] File: /index.html (678) -1
[INFO] File: /js/app.js.gz (210333) -1
[INFO] File: /config.json (346) 1003
[INFO] Starting webserver on port 80
[DEBUG] Checking if web UI is installed in SPIFFS
[DEBUG] Check completed successfully
[INFO] Webserver started, you can now open the user interface on http://192.168.50.132:80/
[INFO] Command Line Interface started, registered 22 commands. Type 'help' to get a list of all supported commands
[INFO] Motion Controller task started
[DEBUG] GET called/api/status
[DEBUG] POST called/api/steppers/moveby with parameters: id=0, value=2000, unit=steps, speed=800, accel=500, decel=500
[DEBUG] GET called/api/status

JuniorGamingTime, 2020-09-15:
This is what is sound like:

https://imgur.com/a/RfC8kbg

pkerspe, 2020-09-15:
ok, serial output seems normal.
If you want the stepper to run less noisy you probably need to enable microstepping, it will reduce oscillations/ripples, at least my stepper runs much quieter in this case. just go for 4 or 8 microsteps setting and see if the noise is less

pkerspe, 2020-09-15:
the sound from the video is a bit strange, maybe you acceleration is to aggressive for your motor

JuniorGamingTime, 2020-09-15:
thats what i thought but even at 100 it makes that noise and at 32 microstepping, so those the other ticking too

JuniorGamingTime, 2020-09-16:
So i just did a really simple test with this code and the nema23 with TB6600 moves smoothly like i expected.
Might be a config error on my part or maybe a bug?

this is what i used:
#define dirPin 32
#define stepPin 33

void setup() {
  // Declare pins as output:
  pinMode(stepPin, OUTPUT);
  pinMode(dirPin, OUTPUT);

  // Set the spinning direction CW/CCW:
  digitalWrite(dirPin, HIGH);
}

void loop() {
  // These four lines result in 1 step:
  digitalWrite(stepPin, HIGH);
  delayMicroseconds(500);
  digitalWrite(stepPin, LOW);
  delayMicroseconds(500);
}

pkerspe, 2020-09-16:
So you are generating a square wave basically with 1khz.
This does not incorporate any acceleration function.
Please try to move the stepper with 1000 steps per second with an acceleration of 500 steps/second and for a distance of let's say 100000 steps.
Once the stepper has reached the speed of 1000 steps per second it should spin for roughly 100 seconds. During this time there is no change in speed so it would be the same as with your 1khz signal (if I go through Math right :-) ) and during this time it should also run at a constant speed with a smooth motion.
Please confirm.
As I understood you only have issues with the strange sound during acceleration phase, which you did not test with your short code snippet (well actually you accelerate with no delay to 1khz, which is also kind of an instant acceleration :-) )

Of course there is always a chance of a bug, yet I cannot reproduce it with my setup here.
Which exact ESP 32 module are you using? Is it the dual core version at all? (There is a single code version available which I never testet and most likely due to the multiple "services" (Tasks) that are started in the ESP-Stepper-Motor-Server, might have some issues with multi tasking)

JuniorGamingTime, 2020-09-16:
The "bump" feeling is not just a sound happens on every rotation. and it still happens 
This is the EPS32 im using: https://www.amazon.com/gp/product/B07Q576VWZ/ref=ppx_yo_dt_b_asin_title_o02_s01?ie=UTF8&psc=1

pkerspe, 2020-09-17:
I am running out of ideas at the moment. Maybe you can try with an older Version of the ESP Stepper Motor Server like Verison 0.2.0?
Also make sure to upload the matching UI version for 0.2.0 (from the examples Folder of this version)

JuniorGamingTime, 2020-09-26:
Sorry been a bit busy, when i try the 0.2.0 version i get this error
src/ESPStepperMotorServer.h:61:26: fatal error: FlexyStepper.h: No such file or directory

pkerspe, 2020-10-09:
that version still uses different depedencies, thus the error. You need to install the FlexyStepper LIbrary (different from ESP_FlexyStepper)

pkerspe, 2020-11-04:
Closing this issue since I cannot reproduce it and no further feedback from TO

closed 2020-11-04 by pkerspe