pkerspe/ESP-StepperMotor-Server #11: 5th motor setting is not visible

savehanges, 2020-09-17:
**Describe the bug**
in the setup page 4 motors are showing properly but when i add 5th motor( even with uploading config.json) setup page only show index id and other variables name but not values. in Control page there are only 4 motor showing but when i download the config file 5th motor is there with all values. 

**Environment**
**IDE: Arduino 1.8.13
ESP_StepperMotor_Server = 0.3.3
ESP-FlexyStepper = 1.4.3
AsyncWebserver = 1.2.3
ArduinoJSON = 6.16.1**

**To Reproduce**
Config.json is attached

**Screenshots**
[image]
[image]

savehanges, 2020-09-17:
`{
    "serverConfiguration": {
        "port": 80,
        "apName": "ESP-StepperMotor-Server",
        "apPassword": "ServerPassword",
        "wififMode": 2,
        "wifiSsid": "",
        "wifiPassword": ""
    },
    "stepperConfigurations": [
        {
            "id": 0,
            "stepPin": 32,
            "directionPin": 33,
            "name": "X-Axis",
            "stepsPerRev": 200,
            "stepsPerMM": 100,
            "microsteppingDivisor": 1,
            "rpmLimit": 1000
        },
        {
            "id": 1,
            "stepPin": 22,
            "directionPin": 23,
            "name": "Y-Axis",
            "stepsPerRev": 200,
            "stepsPerMM": 100,
            "microsteppingDivisor": 1,
            "rpmLimit": 1000
        }
    ],
    "switchConfigurations": [
        {
            "id": 0,
            "ioPin": 14,
            "stepperIndex": 0,
            "switchType": 64,
            "name": "Limit 1 X-Axis",
            "switchPosition": -1
        },
        {
            "id": 1,
            "ioPin": 15,
            "stepperIndex": 1,
            "switchType": 64,
            "name": "Limit 2 X-Axis",
            "switchPosition": 200000
        }
    ],
    "rotaryEncoderConfigurations": [
        {
            "id": 0,
            "pinA": 16,
            "pinB": 17,
            "name": "Encoder X-Axis",
            "stepMultiplier": 100,
            "stepperIndex": 0
        },
        {
            "id": 1,
            "pinA": 18,
            "pinB": 19,
            "name": "Encoder Y-Axis",
            "stepMultiplier": 100,
            "stepperIndex": 1
        }
    ]
}`

pkerspe, 2020-09-17:
this might not be a bug per-se.
To limit the amount of ram used, the number of stepper configurations is limited by a macro in ESPStepperMotorServer.h:
https://github.com/pkerspe/ESP-StepperMotor-Server/blob/586816abc478db28f1900c9ad8234562d2be830e/src/ESPStepperMotorServer.h#L51

You need to define a value here that is number-of-stepper-configs +1 

So you need to put this line at the top of your main.cpp script (hope that works, otherwise need to change ESPStepperMotorServer.h directly):
```#define ESPServerMaxSteppers 10```

savehanges, 2020-09-17:
Thank you for your quick response .. i already defined 6 .. but now again i defined 10 but nothing changed even in control 5th motor is not showing up.
if i remove any motor other then 5th then suddenly its there showing all values in "setup" and "control". 
one more thing i notice. only 5th motor showed up with the "brake settings" with 4 motors  brake settings are not showing.

pkerspe, 2020-09-17:
Thanks for the feedback. I will investigate it as soon as I find the time to do so

pkerspe, 2020-09-21:
the issue should be fixed now, was caused by insufficient memory assigned in the arduino json document (StaticJsonDocument) instance creation. Fixed in Version 0.4.1
please re-test and confirm fix

pkerspe, 2020-09-21:
I also increased the stepper config count to 10 (I doubt anyone will ever need more, since the the ESP32 does not even have enough IO Pins to connect more steppers).
I did NOT test if the movement is still smooth with so many steppers configured

closed 2020-09-21 by pkerspe