pkerspe/ESP-StepperMotor-Server #12: Limit switches in series ?

savehanges, 2020-09-18:
I am little bit confuse on using limit switches in series. when i chose different switch types in "Add a new limit switch configuration" the result in jason file is different then definition in the header file:
https://github.com/pkerspe/ESP-StepperMotor-Server/blob/master/src/ESPStepperMotorServer_PositionSwitch.h#L15
[image]

one other thing i want to know. Is the internalSwitchISR() hard-coded in  staticLimitSwitchISR() with SWITCHTYPE_LIMITSWITCH_POS_BEGIN_BIT ? if yes then its mean i can change it to SWITCHTYPE_LIMITSWITCH_COMBINED_BEGIN_END_BIT and use limit switches in series ?
https://github.com/pkerspe/ESP-StepperMotor-Server/blob/master/src/ESPStepperMotorServer.cpp#L1196

pkerspe, 2020-09-21:
What exactly is your question?

The variable "switchType" is decimal representation of a bit mask.

The bit mask also integrates the information if it is an active high or active low configuration of the switch (SWITCHTYPE_STATE_ACTIVE_HIGH_BIT or SWITCHTYPE_STATE_ACTIVE_LOW_BIT).

So the value decimal 17 is in 10001 in binary, which means the first (decimal 1) and the fifth (decimal 16) bit are set in your configuration: 
first bit indicates SWITCHTYPE_STATE_ACTIVE_HIGH_BIT, so the switch is in active high configuration
the fifth bit indicates SWITCHTYPE_POSITION_SWITCH_BIT, so the switch is a position switch

So by sending the decimal value you configure a position switch in active high configuration.

pkerspe, 2020-09-21:
as for the second part of your post:
yes, it is hardcoded, and yes you can change it to SWITCHTYPE_LIMITSWITCH_POS_BEGIN_BIT
I did not test this at the moment, but in the internalSwitchISR the switch type is evaluated and the proper value should be handed to the ESP_FlexyStepper Library for a combined Start/End switch connected in series.

pkerspe, 2020-09-21:
I pushed a "fix" n version 0.4.1 that actually differentiates between the different limit switch types and thus removed the hardcoded SWITCHTYPE_LIMITSWITCH_POS_BEGIN_BIT

closed 2020-09-21 by pkerspe

pkerspe, 2020-09-21:
in version 0.4.1 I also included the latest UI version now (also 0.4.1) which support selecting the combined (Serial wired) limit switch