pkerspe/ESP-StepperMotor-Server #13: Static IP

Stormer2k0, 2020-10-08:
It would be rather useful to be able to configure a static IP the same way you configure the SSID name and password.
You still can just add a static IP by just by adding this in the main:

` 
IPAddress local_IP(10, 38, 4, 110);
IPAddress gateway(10, 38, 4, 1);
IPAddress subnet(255, 255, 255, 0);

 if (!WiFi.config(local_IP, gateway, subnet)) {
    Serial.println("STA Failed to configure");
  }
`
And if not adding static IP tot he library then maybe add an example for the people who don't know how to do this.

pkerspe, 2020-10-09:
@Stormer2k0 are you sure the syntax is correct?
According to https://www.arduino.cc/en/Reference/WiFiConfig you example is actually the method signature
```WiFi.config(ip, dns, gateway);```

Stormer2k0, 2020-10-09:
I used this syntax as in WiFiSTA.cpp that is the order the function is written: 
[image]

dns1 and dns2 doesn't appear to be nessasary as those don't return a false.

pkerspe, 2020-10-09:
added in version v0.4.2
Example 3 shows how to use it.
```
 IPAddress staticIp(192, 168, 178, 49);
  IPAddress gatewayIp(192, 168, 178, 1);
  IPAddress subnetMask(255, 255, 255, 0);
  stepperMotorServer->setStaticIpAddress(staticIp, gatewayIp, subnetMask);
```

Or you can add the new config nodes to the json config file in the server config section:
```
...
{
  "serverConfiguration": {
...
    "staticIP": "192.168.178.49",
    "gatewayIP": "192.168.178.1",
    "subnetMask": "255.255.255.0"
  },...
```

pkerspe, 2020-10-09:
please test and report if all works as requested

closed 2020-10-09 by pkerspe