pkerspe/ESP-StepperMotor-Server #14: Support for ESP8266

didkivskyy, 2020-11-04:
Hi,
Impressive project.
Is it possible to support boards with **ESP8266** like **NodeMCU** or similar?
They are supported by **platformio** very well.
Or is there anything specific to ESP32 only?

Thanks!

pkerspe, 2020-11-04:
Thanks for your comment.
The ESP32 and the ESP8266 differ quite a lot. Not only do they have different underplaying architecture but also different OS.
I am afraid the current version of the stepper server will not be running without major modifications on the ESP8266.
Libraries would need to be exchanged and the general concept would need to be limited to less functionality.

Main issues might be, that the ESP32 has two cores (except for some models that only come with a single core) which allow for "multitasking" using the underlying FreeRTOS.
Also the ESP32 has more RAM and of course a higher clock Speed.

The stepper server uses a fair amount of the ESP32s Ram if all modules (webserver, CLI, RestAPI) are enables, so that would be a limitation. Also the function to start the server as a service in its own task is not possible on the ESP8266 due to its single core and different "OS".

I can imagine that it is possible to come up with a clone that runs on the ESP8266 by performing the following steps:
- remove the CLI (serial control Interface or refactor it, since currently it start as a separate task)
- Limit the amount of steppers to be controlled (I am just guessing, but not sure if you handle more than 2 or 3 stepper Motors simultaneously)
- remove function to start the server as a service 
- replace the ESPFlexyStepper library with the original FlexyStepper library since this should be still working on a ESP8266 if I remember correctly 
- replace the async webserver library with the one for the ESP8266
- do General refactoring to make everything work with the new libraries

didkivskyy, 2020-11-05:
Yes. Sounds like a lot to do.
So maybe not worth it.

closed 2020-11-06 by pkerspe