pkerspe/ESP-StepperMotor-Server #15: Panic when setting up stepper motor in version 0.4.1

pstoaks, 2021-01-10:
**Describe the bug**
Setting up stepper: (Following is what is displayed on the Setup web page.)
Name: Spindle 
ID: 0
Hardware IO Pins: Step-Pin: 12 / Dir-Pin: 14
Motion-Settings: Steps/mm: 0 / Steps/revolution: 360 Spindle 
                 Microstepping: 2

ll=4 output:
[INFO] Setting log level to ALL
[DEBUG] GET called/api/status
[DEBUG] GET called/api/status
[DEBUG] GET called/api/status
[DEBUG] POST called/api/steppers
[WARNING] Invalid stepper config requested with id 255. Will retun NULL
[DEBUG] GET called/api/steppers
[DEBUG] ArduinoJSON document size uses 808 bytes from alocated 4200 bytes
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400dfcb4  PS      : 0x00060630  A0      : 0x800ea528  A1      : 0x3ffd3740  
A2      : 0x00000000  A3      : 0x3ffdaca8  A4      : 0x3ffc1a38  A5      : 0x5a37194e  
A6      : 0x0000000d  A7      : 0x00000000  A8      : 0x800dfcb1  A9      : 0x3ffd3720  
A10     : 0x3ffd3744  A11     : 0x3f4057f9  A12     : 0x5a37194e  A13     : 0x3ffc1a38  
A14     : 0x00000000  A15     : 0x3ffd375c  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x400dfcb4:0x3ffd3740 0x400ea525:0x3ffd3790 0x400e7ce5:0x3ffd37e0 0x400e7da9:0x3ffd3820 0x400e8011:0x3ffd3860 0x400eb5ad:0x3ffd3880 0x400eb62d:0x3ffd38c0 0x400ebc3e:0x3ffd38e0 0x4008a4a5:0x3ffd3910

**Environment**
 Using Arduino IDE, version 1.8.13

Libraries:
ESP-Stepper-Motor-Server: library.json says 0.4.2 but server says 0.4.1
ESP-Flexy-Stepper: 1.4.3
ArduinoJSON: Rolled back to 6.15.2, but also happens with 6.17.x
ESPAsyncWebServer: 1.2.3
AsyncTCP: 1.1.1

**To Reproduce**
Connect to Web UI
Go to Setup page
Add stepper motor with parameters shown above, then click OK on that form.
See crash in serial monitor.

**Expected behavior**
Stepper motor added, can save config

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
I'm just getting started with it, so nothing special.

pkerspe, 2021-01-10:
could you please decode the backtrace to allow an analysis of the exact line of code where the panic occurs?

pstoaks, 2021-01-10:
C: 0x400dfcb4: std::_Function_handler   >::_M_invoke(const std::_Any_data &,  tmp/arduino_build_173723/Example1_BasicESPStepperMotorServer.ino.elf, CU 0x153dbc, DIE 0x171f5a>) at /home/paul/sandbox/auto/esp8266_thing/libraries/ESP-StepperMotor-Server/src/ESPStepperMotorServer_WebInterface.cpp line 126
EXCVADDR: 0x00000000

Decoding stack results
0x400dfcb4: std::_Function_handler   >::_M_invoke(const std::_Any_data &,  tmp/arduino_build_173723/Example1_BasicESPStepperMotorServer.ino.elf, CU 0x153dbc, DIE 0x171f5a>) at /home/paul/sandbox/auto/esp8266_thing/libraries/ESP-StepperMotor-Server/src/ESPStepperMotorServer_WebInterface.cpp line 126
0x400ea525: AsyncWebServer::AsyncWebServer(unsigned short) at /home/paul/sandbox/auto/esp8266_thing/libraries/ESPAsyncWebServer/src/WebServer.cpp line 52
0x400e7ce5: AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*) at /home/paul/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/xtensa-esp32-elf/include/c++/5.2.0/functional line 2242
0x400e7da9: AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*) at /home/paul/sandbox/auto/esp8266_thing/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 70
0x400e8011: AsyncWebServerRequest::~AsyncWebServerRequest() at /home/paul/sandbox/auto/esp8266_thing/libraries/ESPAsyncWebServer/src/StringArray.h line 162
0x400eb5ad: AsyncClient::_s_connected(void*, void*, signed char) at /home/paul/sandbox/auto/esp8266_thing/libraries/AsyncTCP/src/AsyncTCP.cpp line 1231
0x400eb62d: AsyncClient::_error(signed char) at /home/paul/sandbox/auto/esp8266_thing/libraries/AsyncTCP/src/AsyncTCP.cpp line 866
0x4008a4a5: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

pstoaks, 2021-01-11:
So, the problem was that the stepper icon SVG file wasn't downloaded. I was able to fix this by changing the for loop in ESPStepperMotorServer_WebInterface::checkIfGuiExistsInSpiffs() to loop through all 8 files. No more crash and the config is created and saved properly.

I'm still not getting a legible icon, but I can look into that later

denonbw, 2021-01-12:
> So, the problem was that the stepper icon SVG file wasn't downloaded. I was able to fix this by changing the for loop in ESPStepperMotorServer_WebInterface::checkIfGuiExistsInSpiffs() to loop [...]

Can I ask what exactly you changed?
I have the same problem.

pkerspe, 2021-01-12:
Thanks for the investigation pstoaks.
Interestingly ESP-StepperMotor-Server/src/ESPStepperMotorServer_WebInterface.cpp line 126 this is the request handler for index.html
Did you guys perform the upload step to write the content from the Examples/Example1..../data folder to the SPIFFS?

pkerspe, 2021-01-12:
I just pushed a new commit that performs the complete check over all UI files (the 8 files that pstoaks mentioned) with the version 0.4.3
I could not test it yet though, hope it works

pkerspe, 2021-01-12:
concerning "ESP-Stepper-Motor-Server: library.json says 0.4.2 but server says 0.4.1":
I also bumped the version in the server config to be the same as the release, should be 0.4.3 now also in the server

pkerspe, 2021-01-12:
Just a note to everyone having issues with images not being displayed correctly in the UI:
Please check the log output on the serial port when the server is starting.
At some point you should see a complete listing of all files in the SPIFFS, it should look something like this:
`[INFO] Listing files in root folder of SPIFFS:
[INFO] File: /config.json (883) 48
[INFO] File: /img/rotaryEncoderWheel.svg (13750) -1
[INFO] File: /img/stepper.svg (22739) -1
[INFO] File: /img/emergencyStopSwitch.svg (11442) -1
[INFO] File: /img/switch.svg (19709) -1
[INFO] File: /img/logo.svg (25066) -1
[INFO] File: /upload.html.gz (1212) -1
[INFO] File: /favicon.ico (318) -1
[INFO] File: /index.html (678) -1
[INFO] File: /js/app.js.gz (210386) -1`
All these names have to be listed (as of version 0.4.3 of the stepper motor server with the UI Version 0.4.1, newer versions might have more files listed in the future.
If any of these files is missing, your SPIFFS contents are not fully uploaded.
In this case use the SPIFFS Upload function of your ide (e.g. the function "Upload Filesystem Image" in platformio) to upload the contents of the data folder in the first example provided with the stepper motor server.

pkerspe, 2021-01-13:
in 0.4.4 a selftest page has been added to troubleshoot possible issues with an incomplete set of files in the SPIFFS that are needed for the UI.
Can be called using HTTP://<your esp IP and port>/selftest

denonbw, 2021-01-13:
Hello,
the selftest always gives me "The requested file could not be found.".
But everything else works now. After upload the 0.4.4,  the ESP could download the files:

INFO] The file /img/rotaryEncoderWheel.svg could not be found on SPIFFS
[INFO] Download of /img/rotaryEncoderWheel.svg completed
[INFO] The file /img/emergencyStopSwitch.svg could not be found on SPIFFS
[INFO] Download of /img/emergencyStopSwitch.svg completed
[INFO] The file /img/stepper.svg could not be found on SPIFFS
[INFO] Download of /img/stepper.svg completed
[INFO] The file /img/switch.svg could not be found on SPIFFS
[INFO] Download of /img/switch.svg completed

This was not possible before, although my WLAN is open to the Internet. 
Thanks for the fast and great help!

pkerspe, 2021-01-13:
@denonbw are you sure you removed the "/#" from the regular web UI URL when you tested the selftest?
Or do you see any errors in the console when calling the selftest URL?

denonbw, 2021-01-13:
@pkerspe yes I have tried:
http://IP/selftest and http://IP/api/selftest
No errors in the console.

I loaded all the files fresh from github.
The GUI still shows "connected (UI version: 0.4.1 / Server version: 0.4.3)" Shouldn't that be 0.4.4?

pkerspe, 2021-01-13:
@denonbw can you please check the following location in your current sources:
https://github.com/pkerspe/ESP-StepperMotor-Server/blob/7b7c25f5ad4d53f28cef55ab53a4ed62a99653fc/src/ESPStepperMotorServer.h#L160
This is the line in the code that defines which server version is reported to the frontend. Most likely yours says 0.4.3 (could be still in another position in this file in version 0.4.3, just search the whole file for "version".
In case it says 0.4.3 you did not checkout the latest tagged version v0.4.4

pkerspe, 2021-01-13:
@pstoaks in case your initial issue is closed now, feel free to close this issue

denonbw, 2021-01-13:
> @denonbw can you please check the following location in your current sources:
> https://github.com/pkerspe/ESP-StepperMotor-Server/blob/7b7c25f5ad4d53f28cef55ab53a4ed62a99653fc/src/ESPStepperMotorServer.h#L160
> 
> 
> This is the line in the code that defines which server version is reported to the frontend. Most likely yours says 0.4.3 (could be still in another position in this file in version 0.4.3, just search the whole file for "version".
> In case it says 0.4.3 you did not checkout the latest tagged version v0.4.4, this would also explain why you do not have the selftest page :-)

Sorry that was indeed my mistake!
Is there any way to support your great work?

pkerspe, 2021-01-13:
Uh, support is of course always welcome, feel free to use the sponsor-function on my profile page :-)

pkerspe, 2021-01-14:
Thanks a million @denonbw highly appreciated!

closed 2021-01-14 by pstoaks