pkerspe/ESP-StepperMotor-Server #16: Delay of starting motor after relasing brake

georgekiwi, 2021-01-14:
**Is your feature request related to a problem? Please describe.**
Hi,
I tried your library and I like it a lot!
When I use the stepper without the brake feature, everything works fine.

However, when using the brake feature, the driver does not enable the stepper fast enough, so that about 10-20 steps get lost everytime.

**Describe the solution you'd like**
Is it possible to define an additional variable that controls the time period between _releasing the brake_ and _outputting the motor steps_?

Cheers
gk

pkerspe, 2021-01-14:
@georgekiwi so what you are saying is, that the time between disabling the brake and sending the first step signal is to long, right?
Please provide more details about the speed and acceleration settings you use.
Can you be sure that your stepper has enough tork not to loose steps due to lack of torque?

georgekiwi, 2021-01-14:
Hi,
what I'm saying is that the time is actually too short.
The stepper is strong enough, when I do not use the brake everything works (i.e. one revolution is exactly one revolution). Only when the brake is enabled, one revolution is like 350° or so. This happens with every speed setting.

(I don't know if it is important, but I do not actually use the feature for a brake, I use it to disable the driver so the stepper doesn't get so warm)

pkerspe, 2021-01-14:
I see, thank you for the response.
Did you try sing the "Delay to engage motor brake" and "brake release timeout" settings when creating a new stepper config? Those values are directly feed to the ESP-FlexyStepper instances used

pkerspe, 2021-01-14:
oh, sorry, just got it, these values are just for engaging the brake, not for releasing before moving. Sorry, yeah, that would require quite some work, since it needs to be adapted in the ESP_FlexyStepper library and then in the Server and UI

georgekiwi, 2021-01-15:
Thanks for your response. I'm afraid thats a bit beyond my C-skills ;)

One more thing: The documentation for the CLI has two errors I think
**mt=0&v=100&u=revs** should be **mt=0&v:100&u:revs**
**mb=0&v=-100&u=mm** should be **mb=0&v:-100&u:mm**

Cheers

pkerspe, 2021-01-24:
thanks, I fixed the readme and pushed to master branch