pkerspe/ESP-StepperMotor-Server #17: Iphone (mobile) and Encoder

conaito, 2021-01-20:
HI, i love this solution but have some issues. In my browser the webUI works fine but on my iphone it just view a blank page. can you pls check that out?

Also.. my connected encode seems to not work. on the control view i see no changes if i rotate the encoder. does it need some additional setup?

In this way i would like to ask how the Macro is working or when you release it :)
A great feature also would be if you can add some homedrive and save points for motors to can simple drive to special saved positions - this would be a dream!

BTW i use a M5Stack Core

pkerspe, 2021-01-20:
Hi conaito,
Thank you for your message.
Unfortunately you are opening several topics in one issue, this is hard to follow up over time. May I suggest you open separate tickets for each issue and request?

At the moment I do not have a clear concept yet about the macro function, please feel free to provide specific ideas on the general implementation, functionality and UI design.

As for the two other issues: I haven't tested the incremental/rotary encoder for quite a while to be honest, so there is a chance it is not working anymore. Time wise I am a bit on a tight budget due to other projects and being the only maintainer of this project.

As for the mobile view, I never tested that, the UI has not really been designed for a mobile view, but of course it should at least work somehow. I might have a look into that theses days, but feel free to also provide some more details. It should be possible to connect your iPhone to a Mac (if you have one) and use the developer mode to debug the contexts of mobile safari on your iPhone in the safari browser of your Mac.
If you can provide any error messages form the browser console that would greatly help.

conaito, 2021-01-21:
1. Macro: as i see in the code you already add some functionality. may you can give me a examole to test it? it help to get a good idea for UI

2. Encoder: As i really need the encoder function it would be great if you find a bit time to check it 

3. Iphone: ok i will check if i get any addional debug informations and let you know

conaito, 2021-01-22:
Today i got some new stuff and tested it. Seems that not only Encoder is broken.. also Limit Switch not working. May you can check this? Its really most importnat issue now for me to get it working

pkerspe, 2021-01-22:
Please open separate issues for each problem. Also please provide information about your hardware setup (wiring, IO pins used etc)

pkerspe, 2021-01-24:
closing this issue since it is a duplicate of the newly created, individual tickets

closed 2021-01-24 by pkerspe