pkerspe/ESP-StepperMotor-Server #18: Encoder and Limit Switch

conaito, 2021-01-22:
We use a M5stack core, Gpio 2 and 5 for motors (works well so far). Any limit switch or encoder settings (try on several gpio and other pins) not working. no luck at all

conaito, 2021-01-22:
btw thats may the important part to get it fixed

conaito, 2021-01-24:
any idea about my problem here?

pkerspe, 2021-01-24:
Please post your complete json configuration here

conaito, 2021-01-25:
We use the M5Stack Core and use GPIO Pin 2 and 5 for the motor (works well), GPIO12 for a emergency switch which never will fire (not in console or where ever). Same on Encoder..

thats the config:
{
  "serverConfiguration": {
    "port": 80,
    "wififMode": 2,
    "wifiSsid": "Conaito",
    "wifiPassword": "*****",
    "apName": "@",
    "apPassword": "*****"
  },
  "stepperConfigurations": [
    {
      "id": 0,
      "name": "X-Axis",
      "stepPin": 2,
      "directionPin": 5,
      "stepsPerRev": 200,
      "stepsPerMM": 100,
      "microsteppingDivisor": 8,
      "rpmLimit": 1200
    }
  ],
  "switchConfigurations": [
    {
      "id": 0,
      "name": "Limit 1 X-Axis",
      "ioPin": 12,
      "stepperIndex": 255,
      "switchType": 34,
      "switchPosition": -1
    }
  ],
  "rotaryEncoderConfigurations": [
    {
      "id": 0,
      "name": "Encoder X-Axis",
      "pinA": 13,
      "pinB": 15,
      "stepMultiplier": 100,
      "stepperIndex": 0
    }
  ]
}

here serial console start sequence:

rst:0x1 (POWERON_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[INFO] Setting log level to DEBUG
[DEBUG] Checking SPIFFS for existance and free space
[DEBUG] SPIFFS started
SPIFFS stats:
Total bytes: 1378241
bytes used: 310989
bytes free: 1067252
[INFO] Loading configuration file /config.json from SPIFFS
[INFO] Serializing config 
[DEBUG] File loaded and deserialized
[INFO] 1 stepper configuration entry loaded from config file
[INFO] 1 switch configuration entry loaded from config file
[INFO] 1 rotary encoder configuration entry loaded from config file
[INFO] Serializing config 
{
  "serverConfiguration": {
    "port": 80,
    "wififMode": 2,
    "wifiSsid": "Conaito",
    "wifiPassword": "*****",
    "apName": "@",
    "apPassword": "*****"
  },
  "stepperConfigurations": [
    {
      "id": 0,
      "name": "X-Axis",
      "stepPin": 2,
      "directionPin": 5,
      "stepsPerRev": 200,
      "stepsPerMM": 100,
      "microsteppingDivisor": 8,
      "rpmLimit": 1200
    }
  ],
  "switchConfigurations": [
    {
      "id": 0,
      "name": "Limit 1 X-Axis",
      "ioPin": 12,
      "stepperIndex": 255,
      "switchType": 34,
      "switchPosition": -1
    }
  ],
  "rotaryEncoderConfigurations": [
    {
      "id": 0,
      "name": "Encoder X-Axis",
      "pinA": 13,
      "pinB": 15,
      "stepMultiplier": 100,
      "stepperIndex": 0
    }
  ]
}
[DEBUG] ESPStepperMotorServer_RestAPI instance created
[DEBUG] Motor Controller created
[INFO] Starting ESP-StepperMotor-Server (v. 0.4.4)
[DEBUG] ESPStepperMotorServer compile settings (marcos):
Max steppers: 10
Max switches: 10
Max encoders: 5
[INFO] Trying to connect to WiFi with SSID 'Conaito' ..............
[INFO] Connected to network with IP address 192.168.178.73
[INFO] ESPStepperMotorServer WiFi details:
[INFO] WiFi status: server acts as wifi client in existing network with DHCP
[INFO] SSID: Conaito
[INFO] IP address: 192.168.178.73
[INFO] Strength: -37 dBm
[INFO] Listing files in root folder of SPIFFS:
[INFO] File: /favicon.ico (318) -1
[INFO] File: /img/emergencyStopSwitch.svg (11442) -1
[INFO] File: /img/logo.svg (25066) -1
[INFO] File: /img/rotaryEncoderWheel.svg (13750) -1
[INFO] File: /img/stepper.svg (22739) -1
[INFO] File: /img/switch.svg (19709) -1
[INFO] File: /index.html (680) -1
[INFO] File: /js/app.js.jgz (210386) -1
[INFO] File: /upload.html.gz (1212) -1
[INFO] File: /config.json (539) 1262
[INFO] Starting webserver on port 80
[DEBUG] Checking if web UI is installed in SPIFFS
[DEBUG] Check completed successfully
[INFO] Webserver started, you can now open the user interface on http://192.168.178.73:80/
[DEBUG] Setting up IO pin 12 as input with pullup for active low switch 'Limit 1 X-Axis' (0)
[DEBUG] Setting up IO pin 13 as Pin A input for active high rotary encoder 'Encoder X-Axis' (0)
[DEBUG] Setting up IO pin 15 as Pin B input for active high rotary encoder 'Encoder X-Axis' (0)
[DEBUG] Setting bit 0 to high in register for switch 0 with io pin 12
[DEBUG] Setting bit 1 to low in register for switch 1 with io pin 255
[DEBUG] Attaching interrupt service routine for emergency stop switch 'Limit 1 X-Axis' on IO pin 12
[INFO] Command Line Interface started, registered 22 commands. Type 'help' to get a list of all supported commands
[INFO] Motion Controller task started
[DEBUG] GET called/api/status

conaito, 2021-01-25:
btw: if i try emergencystop in serial console it works well

conaito, 2021-01-26:
i try today a other board and it looks there is soem other bug. please check the limit switch is triggered (Limit 2 Y-Axis), which is correct but as soon i leave the button a wrong Limit Switch (Limit 1 Y-Axis) released. Can you please check the issue and fix that?

attached my config for test:

[INFO] Serializing config 
{
  "serverConfiguration": {
    "port": 80,
    "wififMode": 2,
    "wifiSsid": "Encoder X-Axis",
    "wifiPassword": "*****",
    "apName": "@",
    "apPassword": "*****"
  },
  "stepperConfigurations": [
    {
      "id": 0,
      "name": "X-Axis",
      "stepPin": 15,
      "directionPin": 2,
      "stepsPerRev": 200,
      "stepsPerMM": 100,
      "microsteppingDivisor": 1,
      "rpmLimit": 1000
    },
    {
      "id": 1,
      "name": "Y-Axis",
      "stepPin": 15,
      "directionPin": 2,
      "stepsPerRev": 200,
      "stepsPerMM": 100,
      "microsteppingDivisor": 8,
      "rpmLimit": 1200
    }
  ],
  "switchConfigurations": [
    {
      "id": 0,
      "name": "Limit 1 Y-Axis",
      "ioPin": 0,
      "stepperIndex": 1,
      "switchType": 66,
      "switchPosition": -1
    },
    {
      "id": 1,
      "name": "Limit 2 Y-Axis",
      "ioPin": 35,
      "stepperIndex": 1,
      "switchType": 66,
      "switchPosition": -1
    }
  ],
  "rotaryEncoderConfigurations": [
    {
      "id": 0,
      "name": "Encoder X-Axis",
      "pinA": 16,
      "pinB": 17,
      "stepMultiplier": 100,
      "stepperIndex": 0
    },
    {
      "id": 1,
      "name": "Encoder Y-Axis",
      "pinA": 18,
      "pinB": 19,
      "stepMultiplier": 100,
      "stepperIndex": 1
    }
  ]
}

conaito, 2021-01-26:
additional the emergency as well not working. if i use them on G35 i see follow in serial
[WARNING] Invalid stepper config requested with id 255. Will retun NULL
[INFO] Emergency Switch triggered

pkerspe, 2021-01-26:
are you aware that you configured the step pin for both steppers to be GPIO15?
I never tested something like this and it makes actually no sense, since you would mix up step counters and whatnot.

Please start with a small config: one stepper motor and one emergency switch, not more.
See if you get any errors in the serial console and see if the emergency switch works that way.
I will perform a little test later today with this configuration and report back

conaito, 2021-01-26:
emergency switch works if i set it into the config file and upload it (over gui it set stepper index to 255 which not work then. over manual setup onn file i get them working.

limit switches still have same. please check console (press button switch triggered, on release other switch released):

[INFO] Setting log level to DEBUG
[DEBUG] Checking SPIFFS for existance and free space
[DEBUG] SPIFFS started
SPIFFS stats:
Total bytes: 1378241
bytes used: 310989
bytes free: 1067252
[INFO] Loading configuration file /config.json from SPIFFS
[INFO] Serializing config 
[DEBUG] File loaded and deserialized
[INFO] 1 stepper configuration entry loaded from config file
[INFO] 2 switch configuration entries loaded from config file
[INFO] 0 rotary encoder configuration entries loaded from config file
[INFO] Serializing config 
{
  "serverConfiguration": {
    "port": 80,
    "wififMode": 2,
    "wifiSsid": "Conaito",
    "wifiPassword": "*****",
    "apName": "@",
    "apPassword": "*****"
  },
  "stepperConfigurations": [
    {
      "id": 1,
      "name": "Y-Axis",
      "stepPin": 15,
      "directionPin": 2,
      "stepsPerRev": 200,
      "stepsPerMM": 100,
      "microsteppingDivisor": 8,
      "rpmLimit": 1200
    }
  ],
  "switchConfigurations": [
    {
      "id": 0,
      "name": "Limit 1 Y-Axis",
      "ioPin": 0,
      "stepperIndex": 1,
      "switchType": 66,
      "switchPosition": -1
    },
    {
      "id": 1,
      "name": "Limit 2 Y-Axis",
      "ioPin": 35,
      "stepperIndex": 1,
      "switchType": 66,
      "switchPosition": -1
    }
  ],
  "rotaryEncoderConfigurations": []
}
[DEBUG] ESPStepperMotorServer_RestAPI instance created
[DEBUG] Motor Controller created
[INFO] Starting ESP-StepperMotor-Server (v. 0.4.4)
[DEBUG] ESPStepperMotorServer compile settings (marcos):
Max steppers: 10
Max switches: 10
Max encoders: 5
[INFO] Trying to connect to WiFi with SSID 'Conaito' ............
[INFO] Connected to network with IP address 192.168.178.76
[INFO] ESPStepperMotorServer WiFi details:
[INFO] WiFi status: server acts as wifi client in existing network with DHCP
[INFO] SSID: Conaito
[INFO] IP address: 192.168.178.76
[INFO] Strength: -35 dBm
[INFO] Listing files in root folder of SPIFFS:
[INFO] File: /favicon.ico (318) -1
[INFO] File: /img/emergencyStopSwitch.svg (11442) -1
[INFO] File: /img/logo.svg (25066) -1
[INFO] File: /img/rotaryEncoderWheel.svg (13750) -1
[INFO] File: /img/stepper.svg (22739) -1
[INFO] File: /img/switch.svg (19709) -1
[INFO] File: /index.html (680) -1
[INFO] File: /js/app.js.jgz (210386) -1
[INFO] File: /upload.html.gz (1212) -1
[INFO] File: /config.json (544) 60
[INFO] Starting webserver on port 80
[DEBUG] Checking if web UI is installed in SPIFFS
[DEBUG] Check completed successfully
[INFO] Webserver started, you can now open the user interface on http://192.168.178.76:80/
[DEBUG] Setting up IO pin 0 as input with pullup for active low switch 'Limit 1 Y-Axis' (0)
[WARNING] The configured IO pin 35 cannot be used for active low switches unless an external pull up resistor is in place. The ESP does not provide internal pullups on this IO pin. Make sure you have a pull up resistor in place for the switch d⸮⸮? (1)
[DEBUG] Setting up IO pin 35 as input with pullup for active low switch 'Limit 2 Y-Axis' (1)
[DEBUG] Setting bit 0 to high in register for switch 0 with io pin 0
[DEBUG] Attaching interrupt service routine for limit switch 'Limit 1 Y-Axis' on IO pin 0
[DEBUG] Attaching interrupt service routine for limit switch 'Limit 2 Y-Axis' on IO pin 35
[INFO] Command Line Interface started, registered 22 commands. Type 'help' to get a list of all supported commands
[INFO] Motion Controller task started
[DEBUG] GET called/api/status
[DEBUG] Setting bit 0 to high in register for switch 0 with io pin 0
[DEBUG] Setting bit 1 to low in register for switch 1 with io pin 35
[DEBUG] Limit switch 'Limit 2 Y-Axis' has been triggered (IO pin status is 0)
[DEBUG] Setting bit 0 to high in register for switch 0 with io pin 0
[DEBUG] Limit switch 'Limit 1 Y-Axis' has been released (IO pin status is 1)
[DEBUG] Setting bit 0 to high in register for switch 0 with io pin 0
[DEBUG] Limit switch 'Limit 1 Y-Axis' has been released (IO pin status is 1)
[DEBUG] GET called/api/status

pkerspe, 2021-01-26:
ok, I can replicate the problem with the 255 being set in the config (which is actually fine) but also the warning in the console '[WARNING] Invalid stepper config requested with id 255. Will retun NULL' (which should not happen).
I will work on a fix for that as a starter

pkerspe, 2021-01-26:
BTW: you have seen the warning in your console, right?
`[WARNING] The configured IO pin 35 cannot be used for active low switches unless an external pull up resistor is in place. The ESP does not provide internal pullups on this IO pin. Make sure you have a pull up resistor in place for the switch d⸮⸮?� (1)``
I hope you used external pull up when using GPIO35

conaito, 2021-01-26:
Can you give me a idea what i have to use or how using the pull up resistor? 

i have change to IO PIN 13 but it reboot the board after press:

[INFO] Serializing config 
{
  "serverConfiguration": {
    "port": 80,
    "wififMode": 2,
    "wifiSsid": "Conaito",
    "wifiPassword": "*****",
    "apName": "@",
    "apPassword": "*****"
  },
  "stepperConfigurations": [
    {
      "id": 1,
      "name": "Y-Axis",
      "stepPin": 15,
      "directionPin": 2,
      "stepsPerRev": 200,
      "stepsPerMM": 100,
      "microsteppingDivisor": 8,
      "rpmLimit": 1200
    }
  ],
  "switchConfigurations": [
    {
      "id": 1,
      "name": "Limit 2 Y-Axis",
      "ioPin": 13,
      "stepperIndex": 1,
      "switchType": 66,
      "switchPosition": -1
    },
    {
      "id": 2,
      "name": "stop",
      "ioPin": 35,
      "stepperIndex": 1,
      "switchType": 34,
      "switchPosition": -1
    }
  ],
  "rotaryEncoderConfigurations": []
}
[DEBUG] ESPStepperMotorServer_RestAPI instance created
[DEBUG] Motor Controller created
[INFO] Starting ESP-StepperMotor-Server (v. 0.4.4)
[DEBUG] ESPStepperMotorServer compile settings (marcos):
Max steppers: 10
Max switches: 10
Max encoders: 5
[INFO] Trying to connect to WiFi with SSID 'Conaito' ............
[INFO] Connected to network with IP address 192.168.178.76
[INFO] ESPStepperMotorServer WiFi details:
[INFO] WiFi status: server acts as wifi client in existing network with DHCP
[INFO] SSID: Conaito
[INFO] IP address: 192.168.178.76
[INFO] Strength: -46 dBm
[INFO] Listing files in root folder of SPIFFS:
[INFO] File: /favicon.ico (318) -1
[INFO] File: /img/emergencyStopSwitch.svg (11442) -1
[INFO] File: /img/logo.svg (25066) -1
[INFO] File: /img/rotaryEncoderWheel.svg (13750) -1
[INFO] File: /img/stepper.svg (22739) -1
[INFO] File: /img/switch.svg (19709) -1
[INFO] File: /index.html (680) -1
[INFO] File: /js/app.js.jgz (210386) -1
[INFO] File: /upload.html.gz (1212) -1
[INFO] File: /config.json (539) 41
[INFO] Starting webserver on port 80
[DEBUG] Checking if web UI is installed in SPIFFS
[DEBUG] Check completed successfully
[INFO] Webserver started, you can now open the user interface on http://192.168.178.76:80/
[DEBUG] Setting up IO pin 13 as input with pullup for active low switch 'Limit 2 Y-Axis' (1)
[WARNING] The configured IO pin 35 cannot be used for active low switches unless an external pull up resistor is in place. The ESP does not provide internal pullups on this IO pin. Make sure you have a pull up resistor in place for the switch stop (2)
[DEBUG] Setting up IO pin 35 as input with pullup for active low switch 'stop' (2)
[DEBUG] Setting bit 1 to high in register for switch 1 with io pin 13
[DEBUG] Attaching interrupt service routine for limit switch 'Limit 2 Y-Axis' on IO pin 13
[DEBUG] Setting bit 1 to high in register for switch 1 with io pin 13
[DEBUG] Limit switch 'Limit 2 Y-Axis' has been released (IO pin status is 1)
[DEBUG] Attaching interrupt service routine for emergency stop switch 'stop' on IO pin 35
[INFO] Command Line Interface started, registered 22 commands. Type 'help' to get a list of all supported commands
[INFO] Motion Controller task started
[DEBUG] GET called/api/status
[DEBUG] GET called/api/status
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400810b6  PS      : 0x00060031  A0      : 0x800811ed  A1      : 0x3ffbe750  
A2      : 0x3ffb9974  A3      : 0x00000007  A4      : 0xa5a5a5a5  A5      : 0x80000020  
A6      : 0x3ffc1880  A7      : 0x00000001  A8      : 0x3ffb9578  A9      : 0x3ffbe720  
A10     : 0x000000ff  A11     : 0x000000ff  A12     : 0x3ffb8471  A13     : 0x000000ff  
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xa5a5a5a7  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff8  
Core 1 was running in ISR context:
EPC1    : 0x400810b6  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40086d69

Backtrace: 0x400810b6:0x3ffbe750 0x400811ea:0x3ffbe790 0x4008129d:0x3ffbe7b0 0x40084ab1:0x3ffbe7d0 0x400ef4c2:0x3ffb1fb0 0x40088ebd:0x3ffb1fd0

Rebooting...

pkerspe, 2021-01-26:
You can use a simple 10kOhm Resistor (or something between 1k and 100k, does not matter so much, 10k is just a good start) and connect it between +3,3V and the Pin you connected the switch to. 
Or you just use another GPIO that has an internal pull op available.
See here for pins to use: https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

pkerspe, 2021-01-26:
I just pushed a commit to master branch that fixes the `[WARNING] Invalid stepper config requested with id 255. Will retun NULL`Problem when creating an emergency switch via the GUI

pkerspe, 2021-01-26:
just released a new version 0.4.5 where rotary encoders and emergency switches are working as intended.
General limit switch function not tested yet, will do another day.
I also added a section in the documentation for setting up rotary encoder hardware and how the signals are currently handled. Please read this part if you still have issues with the rotary encoder.

conaito, 2021-01-27:
1. Emergency switch works now as it should (triggered) but if i try any other switch it will not work, also may i not understand the usage - can you explain it?
2. If i add a switch or change it it only update after reset the esp32 - it must reboot. may there is as well a better fix?

{
  "serverConfiguration": {
    "port": 80,
    "wififMode": 2,
    "wifiSsid": "Conaito",
    "wifiPassword": "*****",
    "apName": "@",
    "apPassword": "*****"
  },
  "stepperConfigurations": [
    {
      "id": 0,
      "name": "x",
      "stepPin": 33,
      "directionPin": 23,
      "stepsPerRev": 200,
      "stepsPerMM": 100,
      "microsteppingDivisor": 8,
      "rpmLimit": 1200
    }
  ],
  "switchConfigurations": [
    {
      "id": 0,
      "name": "stop",
      "ioPin": 19,
      "stepperIndex": 255,
      "switchType": 34,
      "switchPosition": -1
    },
    {
      "id": 1,
      "name": "jkhk",
      "ioPin": 22,
      "stepperIndex": 0,
      "switchType": 10,
      "switchPosition": 100
    }
  ],
  "rotaryEncoderConfigurations": []
}

The 2nd switch only view me in console:
[WARNING] Invalid stepper config requested with id 255. Will retun NULL

pkerspe, 2021-01-27:
yes, I noticed that too, that currently a reboot is needed to properly setup the IO Pins for switches etc.
As for the `[WARNING] Invalid stepper config requested with id 255. Will retun NULL`:
please remove the emergency stop switch from your configuration and see if the error still shows. 
Currently, with the config you showed and the fixes provided in Version 0.4.5, this should not show anymore.

pkerspe, 2021-01-27:
I will add a chapter to the manual for the switch types

conaito, 2021-01-27:
i tested the encoder and it works as well like the emergency switch like a charm.
only the other switches i can get working as it may should. maybe you can look into it as well? also the reboot issue to get changes on switches running?

conaito, 2021-01-29:
Hi, may you can find a bit time to take a look into it over the weekend? It becomes a little bit urgent for me and i really want use this solution :/

pkerspe, 2021-01-29:
are you using it in a commercial project or how come this is so urgent?
Maybe you can describe a bit in general what you are trying to achieve in your project, would help to understand which exact needs you have and what might be possible solutions.

conaito, 2021-01-29:
i work on a lift where i need a good start solution to handle the stepper motor and found your project - it looks like to be exactly what i looking for except the bugs i reported. all my mechanical parts mounted here and wait for a fix of software :) thats i mean with little bit urgent

conaito, 2021-02-01:
May i also can try to fix it but it would be great if you can give me a idea where i can start or where it may can happened. Since the Emergency stop and encoder was fixed very fast i think its not a real big issue?

conaito, 2021-02-08:
i found a nearly working solution but have still a issue on the serial switches. it works so far but only if i have a emergency switch (if i delete it the esp will reboot on trigger one of the serial switches). also problem is too that the serial switches trigger on press it and as well on release it - it means it will not work because if stepper move in other direction and i release the switch, the stepper stops again. any idea how i can fix this?

closed 2021-02-09 by conaito

pkerspe, 2021-02-09:
It sounds like you need to debounce your switches if you get false triggers

conaito, 2021-02-09:
yes looks like but you may have a idea how i can do it?

pkerspe, 2021-02-09:
how about using google on the term "debounce switch"? :-) It is a general electrical term not anything specific to the ESP or this library. It also depends what kind of switches you actually use if you need to debounce or not.

conaito, 2021-02-09:
i was think that you may have it in mind since you made the stepper server project. is it not already implemented by using limited switches?

pkerspe, 2021-02-09:
since I do not know what kind of switches you use, I cannot provide a simple fits-all solution. Debouncing circuits need to be designed to match the switch, otherwise they might not be effective or cause to long delays before triggering

conaito, 2021-02-09:
i just use simple micro limit switches in serial. they working well so far but just must bounce back to trigger real witch i not have a idea to make it :/

pkerspe, 2021-02-09:
oh, you mean your switches only "fire" once you release them? Then you wired them up wrong. They need to fire when pressed of course. Most micro switches have NC and NO pinouts, so you can wire them both ways. If wired in serial, it is best practice to use the NC (normally closed) pins since it will also trigger the switches when a wire breaks completely.
In the ESP Stepper Motor server you would configured the switches as active low switches then and you connect one the switches to the IO PIN with one wire and the other wire goes to the supply voltage (e.g. 3.3V).

conaito, 2021-02-09:
oh i made it exactly as you say but 2nd wire to GND (not voltage) may thats the problem??

pkerspe, 2021-02-09:
no you are right, I was wrong, just fixed it in my previous post. To GND is correct

conaito, 2021-02-09:
so i still have the problem :( its as follow: 1 of both switches stop the motor as it should (and good that he can only move in other direction) but i real live then the switch release and stop motor again as soon its open the switch

pkerspe, 2021-02-09:
yeah, sounds like a debouncing problem to me.
Try to press the switch manually, then the motor stops. 
Hold down the switch while sending the motor into the other direction.
Now slowly open the switch (the slower the less chance of debouncing usually, but no guarantee :-) ) and see if the motors stops again. If it does not stop, it is a clear debouncing problem.

conaito, 2021-02-09:
i have try it. as soon the switch is open on move in other direction the motor stop again - thats what i mean. in real it means for me it can’t work :/

pkerspe, 2021-02-09:
it is probably easier to use non mechanical switches, like hall effect sensors or switches that are bounce free from the manufacturer, if you are not into electronics. To effectively calculate a hardware denounce circuit (you need to calculate resistor sizes to match a capacitor size to calculate the correct denounce period for your specific switch, which is much easier if you have a oscilloscope at hand to measure the bounce length) the does not delay the signal to much so that the motor overshoots completely at fast movements is not a trivial thing.
Software debouncing could be implemented rather simply in a dumb program that only denounces one switch and does not need to be hyper efficient, but in the ESP StepperMotor the amount of steppers connected is not fixed thus it makes the tasks much more complex, an even in software you need to set the correct denounce delay to not delay the signal for to long.

pkerspe, 2021-02-09:
hm, I cannot reproduce this unfortunately, maye your wiring is somewhere wrong