pkerspe/ESP-StepperMotor-Server #20: Mobile

conaito, 2021-01-22:
We have test the gui on several browsers.. work on all except on a mobile (iphone se). any idea about? we also try to recompile the gui (vue) but all we try it doesn‘t fix that. we will try soon to provide some debug info iphone/mac

pkerspe, 2021-01-24:
It seems there is an issue with the js file.
When connecting the iPhone to the Mac and inspecting the console in safari with the web developer feature enabled on the iPhone I can see:
```SyntaxError: Invalid character '\u001f'````

conaito, 2021-01-24:
as i read may safari have some problem with gz extension.. so someone talks about using jgz as extension. i also found this code: https://github.com/vitotai/BrewPiLess/blob/master/src/BrewPiLess.cpp

scroll to   //workaround for safari

may thats a idea?

pkerspe, 2021-01-24:
I don't know form the code how this "jgz" file is created, so unfortunately that does not help

pkerspe, 2021-01-24:
ok, in regular safrai it could be fixed by patching the WebResponses.cpp in ESP Async WebServer:
line 534
`    snprintf(buf, sizeof (buf), "inline; filename=\"%s\"", filename);``
needs to be changed to:
`    snprintf(buf, sizeof (buf), "inline");``

pkerspe, 2021-01-24:
I created an issue in ESP Async Webserver Project for this matter:
https://github.com/me-no-dev/ESPAsyncWebServer/issues/926

closed 2021-01-24 by pkerspe

conaito, 2021-01-25:
I found another fix. changing into the ESPStepperMotorServer_WebInterface.h the const char *webUiJsFile = "/js/app.js.jgz"; (Extension jgz) and as well into the data folders the app.js.gz to .jgz (+ into index.html) it also work great now into iphone. :)

pkerspe, 2021-01-25:
did you also test in other browsers if it still works there? 
Not sure if this is a fix or a workaround, but good for you it works!
I will way for the external bug to be closed since this seems more appropriate, but good to know, always learning something new.

conaito, 2021-01-25:
yes tested in several browsers and works well :) hope that you fast can help us for the switch/encoder issue - must say love the solution and not want change to others