pkerspe/ESP-StepperMotor-Server #21: Server Doesnt Show

Humanoidx, 2021-01-22:
**Describe the bug**
After Installation,  the server doesnt show the home page, but the test page and the OTA page show fine.
**Environment**
PlatformIO:

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino

lib_deps = ESP-StepperMotor-Server

monitor_speed = 115200

**To Reproduce**
Uses the above Platform.ini
and Default sketch
Builds fine, Uploads Fine
Says server is running on 192.168.1.20
but when you go there on multiple browser with cache off, just a blank page
The SelfTest and OTA pages display fine,

**Expected behavior**

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
> Executing task in folder ESP STEPPER SERVER: C:\Users\HumanoidX\.platformio\penv\Scripts\pio.exe device monitor --environment esp32dev <

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on COM3  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

rst:0x1 ([INFO] Setting log level to INFO
[WARNING] Failed to load configuration file from SPIFFS. File /config.json not found or SPIFFS is not enabled/mounted
[INFO] Starting ESP-StepperMotor-Server (v. 0.4.4)        
[INFO] Trying to connect to WiFi with SSID 'XPTPCREWX' .....
[INFO] Connected to network with IP address 192.168.1.20
[INFO] Listing files in root folder of SPIFFS:
[INFO] File: /index.html (678) -1
[INFO] File: /js/app.js.gz (29440) -1
[INFO] File: /img/logo.svg (25066) -1
[INFO] File: /favicon.ico (318) -1
[INFO] File: /img/rotaryEncoderWheel.svg (13750) -1
[INFO] File: /img/emergencyStopSwitch.svg (11442) -1
[INFO] File: /img/stepper.svg (22739) -1
[INFO] File: /img/switch.svg (19709) -1
[INFO] File: /upload.html.gz (1212) -1
[INFO] Starting webserver on port 80
[INFO] Webserver started, you can now open the user interface on http://192.168.1.20:80/
[INFO] Command Line Interface started, registered 22 commands. Type 'help' to get a list of all supported commands
[INFO] Motion Controller task started

pkerspe, 2021-01-22:
Can you please open the console of your browser (e.g. by pressing F12 in chrome) and check of there are some errors showing?
Also please check if there is any source code loaded or if it is basically a blank response from the server when you open the root path / home page

Humanoidx, 2021-01-22:
Under Console it says :
`  Uncaught SyntaxError: Unexpected end of input ... app.js.gz:25`

sources shows:
the js folder
the app.js.gz file with:
```
(function(t){var e={};function o(r){if(e[r])return e[r].exports;var n=e[r]={i:r,l:!1,exports:{}};return t[r].call(n.exports,n,n.exports,o),n.l=!0,n.exports}o.m=t,o.c=e,o.d=function(t,e,r){o.o(t,e)||Object.defineProperty(t,e,{enumerable:!0,get:r})},o.r=function(t){"undefined"!==typeof Symbol&&Symbol.toStringTag&&Object.defineProperty(t,Symbol.toStringTag,{value:"Module"}),Object.defineProperty(t,"__esModule",{value:!0})},o.t=function(t,e){if(1&e&&(t=o(t)),8&e)return t;if(4&e&&"object"===typeof t&&t&&t.__esModule)return t;var r=Object.create(null);if(o.r(r),Object.defineProperty(r,"default",{enumerable:!0,value:t}),2&e&&"string"!=typeof t)for(var n in t)o.d(r,n,function(e){return t[e]}.bind(null,n));return r},o.n=function(t){var e=t&&t.__esModule?function(){return t["default"]}:function(){return t};return o.d(e,"a",e),e},o.o=function(t,e){return Object.prototype.hasOwnProperty.call(t,e)},o.p="/",o(o.s=0)})({0:function(t,e,o){t.exports=o("56d7")},"034f":function(t,e,o){"use strict";var r=o("3ebb"),n=o.n(r);n.a},"07d6":function(t,e){t.exports=function(){throw new Error("define cannot be used indirect")}},"09cb":function(t,e,o){var r=o("1918");"string"===typeof r&&(r=[[t.i,r,""]]),r.locals&&(t.exports=r.locals);var n=o("499e").default;n("41f4ec7f",r,!0,{sourceMap:!1,shadowMode:!1})},"0a06":function(t,e,o){"use strict";var r=o("c532"),n=o("30b5"),i=o("f6b4"),a=o("5270"),s=o("4a7b");function l(t){this.defaults=t,this.interceptors={request:new i,response:new i}}l.prototype.request=function(t){"string"===typeof t?(t=arguments[1]||{},t.url=arguments[0]):t=t||{},t=s(this.defaults,t),t.method?t.method=t.method.toLowerCase():this.defaults.method?t.method=this.defaults.method.toLowerCase():t.method="get";var e=[a,void 0],o=Promise.resolve(t);this.interceptors.request.forEach((function(t){e.unshift(t.fulfilled,t.rejected)})),this.interceptors.response.forEach((function(t){e.push(t.fulfilled,t.rejected)}));while(e.length)o=o.then(e.shift(),e.shift());return o},l.prototype.getUri=function(t){return t=s(this.defaults,t),n(t.url,t.params,t.paramsSerializer).replace(/^\?/,"")},r.forEach(["delete","get","head","options"],(function(t){l.prototype[t]=function(e,o){return this.request(r.merge(o||{},{method:t,url:e}))}})),r.forEach(["post","put","patch"],(function(t){l.prototype[t]=function(e,o,n){return this.request(r.merge(n||{},{method:t,url:e,data:o}))}})),t.exports=l},"0df6":function(t,e,o){"use strict";t.exports=function(t){return function(e){return t.apply(null,e)}}},1157:function(t,e,o){var r,n;
/*!
 * jQuery JavaScript Library v3.4.0
 * https://jquery.com/
 *
 * Includes Sizzle.js
 * https://sizzlejs.com/
 *
 * Copyright JS Foundation and other contributors
 * Released under the MIT license
 * https://jquery.org/license
 *
 * Date: 2019-04-10T19:48Z
 */(function(e,o){"use strict";"object"===typeof t.exports?t.exports=e.document?o(e,!0):function(t){if(!t.document)throw new Error("jQuery requires a window with a document");return o(t)}:o(e)})("undefined"!==typeof window?window:this,(function(o,i){"use strict";var a=[],s=o.document,l=Object.getPrototypeOf,c=a.slice,d=a.concat,p=a.push,u=a.indexOf,b={},f=b.toString,m=b.hasOwnProperty,g=m.toString,h=g.call(Object),v={},x=function(t){return"function"===typeof t&&"number"!==typeof t.nodeType},w=function(t){return null!=t&&t===t.window},y={type:!0,src:!0,nonce:!0,noModule:!0};function k(t,e,o){o=o||s;var r,n,i=o.createElement("script");if(i.text=t,e)for(r in y)n=e[r]||e.getAttribute&&e.getAttribute(r),n&&i.setAttribute(r,n);o.head.appendChild(i).parentNode.removeChild(i)}function C(t){return null==t?t+"":"object"===typeof t||"function"===typeof t?b[f.call(t)]||"object":typeof t}var A="3.4.0",S=function(t,e){return new S.fn.init(t,e)},_=/^[\s\uFEFF\xA0]+|[\s\uFEFF\xA0]+$/g;function T(t){var e=!!t&&"length"in t&&t.length,o=C(t);return!x(t)&&!w(t)&&("array"===o||0===e||"number"===typeof e&&e>0&&e-1 in t)}S.fn=S.prototype={jquery:A,constructor:S,length:0,toArray:function(){return c.call(this)},get:function(t){return null==t?c.call(this):t<0?this[t+this.length]:this[t]},pushStack:function(t){var e=S.merge(this.constructor(),t);return e.prevObject=this,e},each:function(t){return S.each(this,t)},map:function(t){return this.pushStack(S.map(this,(function(e,o){return t.call(e,o,e)})))},slice:function(){return this.pushStack(c.apply(this,arguments))},first:function(){return this.eq(0)},last:function(){return this.eq(-1)},eq:function(t){var e=this.length,o=+t+(t<0?e:0);return this.pushStack(o>=0&&o<e?[this[o]]:[])},end:function(){return this.prevObject||this.constructor()},push:p,sort:a.sort,splice:a.splice},S.extend=S.fn.extend=function(){var t,e,o,r,n,i,a=arguments[0]||{},s=1,l=arguments.length,c=!1;for("boolean"===typeof a&&(c=a,a=arguments[s]||{},s++),"object"===typeof a||x(a)||(a={}),s===l&&(a=this,s--);s<l;s++)if(null!=(t=arguments[s]))for(e in t)r=t[e],"__proto__"!==e&&a!==r&&(c&&r&&(S.isPlainObject(r)||(n=Array.isArray(r)))?(o=a[e],i=n&&!Array.isArray(o)?[]:n||S.isPlainObject(o)?o:{},n=!1,a[e]=S.extend(c,i,r)):void 0!==r&&(a[e]=r));return a},S.extend({expando:"jQuery"+(A+Math.random()).replace(/\D/g,""),isReady:!0,error:function(t){throw new Error(t)},noop:function(){},isPlainObject:function(t){var e,o;return!(!t||"[object Object]"!==f.call(t))&&(e=l(t),!e||(o=m.call(e,"constructor")&&e.constructor,"function"===typeof o&&g.cal [... cut ...]
            window.serverUIVersion = "0.4.1";
        </script>
        <link href=/js/app.js.gz rel=preload as=script>
    </head>
    <body class=h-100>
        <noscript>
            <strong>We're sorry but esp-steppermotor-server-ui doesn't work properly without JavaScript enabled. Please enable it to continue.</strong>
        </noscript>
        <div class=container-fluid>
            <div id=app class=row></div>
        </div>
        <script src=/js/app.js.gz></script>
    </body>
</html>

```

pkerspe, 2021-01-22:
Ok, so your app.js.gz file on the ESP32 seems to be corrupt.
It is cut off at the end.
Please compare the end of the file with the end of the file on your hard disk that you uploaded to spiffs.
Maybe just try uploading spiffs contents again and check for errors in the console while uploading.

pkerspe, 2021-01-24:
I just uploaded UI version 0.4.2 to SPIFFS and installed server version 0.4.4
I cannot reproduce the issue, all works fine in Chrome browser (not tested in other browsers though)
Please try uploading the data folder again to SPIFFs.

The app.js file should end with:
```
[...]shadowMode:!1})}});
//# sourceMappingURL=app.js.map
````

Humanoidx, 2021-01-24:
Ok I had to delete all my PlatformIO IDE and reinstall it and then in worked!
Thank you much for your support.

closed 2021-01-25 by pkerspe