pkerspe/ESP-StepperMotor-Server #22: Limit Switches

conaito, 2021-02-05:
Its absolutly not understandable. i added a emergency switch.. works well. added a end switch.. works as well.. if i then add a home switch i get always follow debug message: "[WARNING] A IO Pin change has been detected for switch id 3 which is not a limit switch, but the ISR was triggered for a switch of type limit switch. It is possible that a limit switch status change has not been detected properly" and in this way both end and home switch trigger it.

here my config:
"stepperConfigurations": [
    {
      "id": 0,
      "name": "X Achse",
      "stepPin": 2,
      "directionPin": 15,
      "stepsPerRev": 200,
      "stepsPerMM": 100,
      "microsteppingDivisor": 8,
      "rpmLimit": 1200
    }
  ],
  "switchConfigurations": [
    {
      "id": 0,
      "name": "Panic Stop",
      "ioPin": 13,
      "stepperIndex": 0,
      "switchType": 34,
      "switchPosition": -1
    },
    {
      "id": 1,
      "name": "end",
      "ioPin": 12,
      "stepperIndex": 0,
      "switchType": 10,
      "switchPosition": -1
    },
    {
      "id": 2,
      "name": "home",
      "ioPin": 26,
      "stepperIndex": 0,
      "switchType": 10,
      "switchPosition": -1
    }
  ],
  "rotaryEncoderConfigurations": []
}

conaito, 2021-02-05:
btw i tested same connections and setup with FlexyStepper direct and there is no problem.

closed 2021-02-08 by conaito

BenSpex, 2021-08-13:
Hi @conaito How did you solve this?

I want to use the limit switches to set a home position. Eg. On boot move into one direction until limit switch is reached and set this position as home (0) 
However setting the limit switch I get the exact same error message as you do. 

My basic setup would be:

Start Move Until LimitSwitch has been reached, set this position as 0. Do some movements. Move back to Home and reset home position in case some steps got lost along the way. 

But I do not seem to find any of that functionality. Besides the set home macro. 

Thanks 
Ben