pkerspe/ESP-StepperMotor-Server #24: ui files not exist

dibg, 2021-02-15:
**Describe the bug**
uploaded the application, connected to a wifi with internet connectio and a dhcp server that will provide a valid connection but the esp32 never download the files.
I tried /update noting.

/selftest:
ESP-StepperMotorServer self test

Testing environment:

    Server Version: 0.4.5
    SPIFFS Initialized: true
    WEB UI installed completely: true
    Listing files in root folder of SPIFFS:
        File: //. (0) 218
        File: /index.html.gz (79777) 302
        File: /index.html (0) -1
        File: /js/app.js.gz (0) -1
        File: /img/logo.svg (0) -1
        File: /favicon.ico (0) -1
        File: /img/rotaryEncoderWheel.svg (0) -1
        File: /img/emergencyStopSwitch.svg (0) -1
        File: /img/stepper.svg (0) -1
        File: /img/switch.svg (0) -1
        File: /upload.html.gz (0) -1

{"version":"0.4.5","wifi":{"mode":"client","ip":"192.168.2.3"},"spiffss":{"total_space":1378241,"free_space":1291395},"activeModules":{"serial_cli":true,"rest_api":true,"web_ui":true}}
**Environment**
I tried both Arduino and PlatformIO under manajaro. The same result with both. 
ESP-StepperMotor-Server 0.4.5
ESP-FlexyStepper 1.4.1
AsyncWebserver 1.2.3
ArduinoJSON 6.15.2

**To Reproduce**
Upload the program to the esp32.
Go to the link the serial monitor returns.
see what the esp return, in my case empty page.

**Expected behavior**
in my case empty page.

**Additional context**
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6364
entry 0x400806b8
[INFO] Setting log level to INFO
[WARNING] Failed to load configuration file from SPIFFS. File /config.json not found or SPIFFS is not enabled/mounted
[INFO] Starting ESP-StepperMotor-Server (v. 0.4.5)
[INFO] Trying to connect to WiFi with SSID 'Node 2' ....
[INFO] Connected to network with IP address 192.168.2.3
[INFO] Listing files in root folder of SPIFFS:
[INFO] File: //. (0) 218
[INFO] File: /index.html.gz (79777) 302
[INFO] File: /index.html (0) -1
[INFO] File: /js/app.js.gz (0) -1
[INFO] File: /img/logo.svg (0) -1
[INFO] File: /favicon.ico (0) -1
[INFO] File: /img/rotaryEncoderWheel.svg (0) -1
[INFO] File: /img/emergencyStopSwitch.svg (0) -1
[INFO] File: /img/stepper.svg (0) -1
[INFO] File: /img/switch.svg (0) -1
[INFO] File: /upload.html.gz (0) -1
[INFO] Starting webserver on port 80
[INFO] Webserver started, you can now open the user interface on http://192.168.2.3:80/
[INFO] Command Line Interface started, registered 22 commands. Type 'help' to get a list of all supported commands
[INFO] Motion Controller task started

dibg, 2021-02-15:
(facepalm) i thought that the files will automatically pulled from the repository but that was not the case.
Putting the data folder from https://github.com/pkerspe/ESP-StepperMotor-Server-UI in the root directory of the project will create the image and upload it to the chip. 
You can probably clarify this on the read me if needed.

pkerspe, 2021-02-15:
Hi @dibg ,

Thank you for you report.
Manually Uploading the UI is described in the readme file already.
The auto download alternative seems to kick in on your ESP bit did not complete for whatever reason.
The output on the test page showed that all files existed but most where with a size of zero (that of course is not good).
Unfortunately you did not provide the output of the first boot where the auto download function kicked in.
I will try to add a zero-Filesize-Check though, to prevent creating empty files.

dibg, 2021-02-20:
yes the zero file size seems a good idea, should i close this issue ?

closed 2021-04-24 by dibg