pkerspe/ESP-StepperMotor-Server #25: ESP IO pin for motor brake is not saved

dibg, 2021-02-20:
**Describe the bug**
 "Save current configuration to device" button doesn't save the "ESP IO pin for motor brake" setting. On the next boot this must be re entered.

**Environment**
the same with : https://github.com/pkerspe/ESP-StepperMotor-Server/issues/24

denonbw, 2022-02-11:
Sadly I have the same problem.
As an interim solution you can simply create the stepper via Config and define the pin there.

`void setup() 
{
  Serial.begin(115200);
  stepperMotorServer = new ESPStepperMotorServer(ESPServerRestApiEnabled | ESPServerWebserverEnabled | ESPServerSerialEnabled);
  stepperMotorServer->setWifiCredentials(wifiName, wifiSecret);
  stepperMotorServer->setWifiMode(ESPServerWifiModeClient); //start the server as a wifi client (DHCP client of an existing wifi network)

  stepperConfiguration = new ESPStepperMotorServer_StepperConfiguration(STEP_PIN, DIRECTION_PIN);
  stepperConfiguration->setDisplayName("M1");
  stepperConfiguration->setMicrostepsPerStep(1);
  stepperConfiguration->setBrakeIoPin(5, 1);
  stepperConfiguration->setStepsPerMM(200);
  stepperConfiguration->setStepsPerRev(800);
  stepperMotorServer->addOrUpdateStepper(stepperConfiguration);
  
  stepperMotorServer->start();
}`

pkerspe, 2022-03-07:
Currently adding the following details to the configuration persistence logic:
```
nestedStepperConfig["breakPin"] = stepperConfig->getBrakeIoPin();
nestedStepperConfig["breakPinActiveState"] = stepperConfig->getBrakePinActiveState();
nestedStepperConfig["breakEngageDelay"] = stepperConfig->getBrakeEngageDelayMs();
nestedStepperConfig["breakReleaseDelay"] = stepperConfig->getBrakeReleaseDelayMs();
```

pkerspe, 2022-03-19:
Has been fixed in version v0.4.8

closed 2022-03-19 by pkerspe