pkerspe/ESP-StepperMotor-Server #26: Switches connected to ESP32

une18, 2021-03-22:
How to connect button or switch to ESP32 for start, stop, left, right? 

Is it possible to create a simple row of commands 1000 steps left, delay 2000ms, 2000 steps right, delay 1000ms....

pkerspe, 2021-03-22:
Hi Une18, This is a software library, not a hardware connectivity support forum :-)

closed 2021-03-22 by pkerspe

une18, 2021-03-23:
I know how to wire the buttons, I did not find a way how to tell the software what one or the other button conneted to any GPIO pins should do.

I asked, for the switches, because I could not find information about it in the documentation. Still can not...

I wanted to know if it is possible to do all this in UI, I can not write code myself.

Will order another encoder. Would you reccommend a specific model?

pkerspe, 2021-03-23:
Other than limit switch functionality and rotary encoders, additional switches and functionality are currently not supported to be created via the UI.
The features you require need custom c++ code to be written.
You are looking for the Marco-Functionality which is planned for some future release but not implemented yet. This feature will still take some time. See #19