pkerspe/ESP-StepperMotor-Server #27: Guidance to chain multiple esp together for LEGO toy.

arturohernandez10, 2021-04-26:
I want to build a lego stepper system programmed using scratch for my son. Ease of wiring would be ideal. My current thought is to include an esp in every motor block. And have a USB or similar cable connecting blocks and batteries together. Any motor could be powered by connecting to a powered motor or a battery pack. There is enough learning for my son just with scratch, I want the motors to be as much plug and play as possible.

The exciting part of this. Is that most of the hard work is already done by you and the scratch group. In case you are not familiar with scratch here is a link https://scratch.mit.edu . And that this lego motor system would be much better than the commercial version for a fraction of the price.

I realize there is a huge number of ways to communicate between ESP devices. My current thought is to have an esp at the battery as the bridge to the scratch browser app. This esp would send the message to all connected motors, and each motor would either recognize themselves as the recipient and execute it. Or they would send it down to the next motors. 

Do you have any suggestions or guidance of how to implement this chaining feature?

In any case thank you for your really cool libraries. I've gone through the docs before but this will be the first time I get to use them.

pkerspe, 2021-04-27:
Hello @arturohernandez10,
thank you for your post, yet I am not sure how this relates really to this project. 
It seems you are facing some general design challenges/decisions for inter MCU communication and as you said there is a myriad of possibilities to solve this requirement.
Even though I do not see much connection to this specific library, other than controlling a stepper driver, I cab give you the following input:
- if all you want to do is control a single stepper motor driver, you do not need a power hungry and "expensive" ESP32, also since the ESP32 does not have any native USB hardware it might not be such a good choice for this project, at least when it comes to real USB communication
- do not forget that you also need to include a stepper driver in your application and also that only small stepper motors run of such low voltages as 3.3V that the ESP requires. Plus they consume quite some power, nut sure about the size of battery you are planning to use, combined with the power hunger of the ESP32 (with wifi you go easily above the 250ma range, at least during transmission) this might drain a battery rather quick.
-- Which Stepper motor and drive do you want to use for this project?
I assume that a small overall package size is required so you need to fit the ESP32, the driver and the stepper all together in a small package. 

So if I would do something like this, I would go for a smaller MCU tailored to the requirements and stripping away all unneeded features. For example, you could use an ATMEGA32U4 (just the first to come to my mind) which has native USB Support and consumers way less power...well It does not have any radio (Wifi, Bluetooth) but you would not need that with a wired connection anyway if you go that way.

Another way would of course be to use some radio signals to communicate between the devices (if you go for e.g. with the ESP32): then I would go for BLE.

If you want to use wired communication than you could go for I2C (ATTENTION: total bus length would be limited with this interface to probably less than 1 meter if you do not want to go down on the data rate too much). Most of the MCUs on the market do offer I2C interfaces (implemented in hardware) or you can always bitbang it.

If you want to cover longer ranges you can use RS232 (Serial), you can get boards with interface chips like the MAX3232 for some 10-20 cents when you buy them from china on the bay (https://www.ebay.de/itm/310649984062?hash=item48542e943e:g:YnsAAOSw6khazJH2). You would just need to implement your own serial protocol to communicate between multiple devices.
Alternatively you could also go for RS485 for even longer ranges and more noise immunity.

So some decisions you need to come up with first:
- what battery will you use? (Voltage and Capacity)
- Which stepper motor do you plan to use
- which stepper driver do you plan to use or do you maybe want to build you own one to reduce size?
- which communication interface to use? (wired or RF)
- what distances should be bridged between the different MCUs?
- how much computing power do you really need? Are there other tasks to be performed than just putting out pulses to a stepper driver?
- do you have any size limitations for the whole package?

If it was for me I would do the following from what I understand so far in regards to you requirements:
- create my own PCB with a Microchop MCU like the above mentioned ATMEGA32U4
- implement a stepper driver circuit on the same PCB (following one of the many A4988 style module you get for small money online)
- add a MAX3232 or similar RS232 to TTL converter Chip
- chose a stepper motor that runs of 3,7 Volts (or less)

The PCB would be the same size as the stepper motor back (E.g. NEMA17) so I could "piggy back" it on the steper motor and basically would not increase the overall dimensions other than the depth of the stepper motor.
Reasonable capacity battery could also be piggy backed on it, but then you would also need to add some charging IC to prevent over- and undercharge, or you power the whole thing externally (probably better anyway and just add a voltage regulator to the PCB just to be sure to protect the MUC from overcurrent.

arturohernandez10, 2021-05-10:
This is such great help!! I was thinking maybe an atmega would be a better option. I see what you say about not being so related to your project. The scratch application runs 100% on the browser. So I was thinking, I could store the static files on a server and have the esp32 be a passthrough for those files. So that I could make API calls from new Scratch components, directly to your API perhaps with some changes. But that ignores the protocol issue since I say I would have one MCU per motor.

I have researched I2C and without knowing the details of USB I think it may be a simpler choice. But I'll have to do more homework on that.

But since you were so kind to give me such a thorough answer. I have this conundrum to share with you. I would love to follow your advice to the T. But I'm not sure if I would be signing up for too much work. Even when I know my way on MCUs. My day job is just software. Still you laid out the best plan for the project. My instinct tells me I may find a ready made board with the chips you mention already put together. And I may end up trying the pcb route. I know nowadays. It's so easy to order them online.

Please feel free to close the issue if you like.

pkerspe, 2021-05-10:
for starter / proof of concept you do not need to go the route of your own PCB just yet, you can just buy the ATMEGA32U4, stepper driver and maybe serial driver board separately and wire them together on a breadboard.
Here are some ebay links to get started:

ATMEGA32U4 Board:
https://www.ebay.com/itm/312127537165

Small Stepper Driver (rated up to 1.5A without cooling):
https://www.ebay.com/itm/202171650433?hash=item2f125e9181:g:dA0AAOSwETJaTJqp

Serial to RS232 Adapter:
https://www.ebay.com/itm/310649984062?hash=item48542e943e:g:YnsAAOSw6khazJH2

pkerspe, 2021-05-10:
closing this issue now but feel free to continue the exchange here if needed. Just closing since it is not directly related to the library.

closed 2021-05-10 by pkerspe