pkerspe/ESP-StepperMotor-Server #30: blank web Ui after flashing config

FigJam23, 2021-12-18:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
E (144) psram: PSRAM ID read error: 0xffffffff
[INFO] Setting log level to INFO
[WARNING] Failed to load configuration file from SPIFFS. File /config.json not found or SPIFFS is not enabled/mounted
[INFO] Starting ESP-StepperMotor-Server (v. 0.4.6)
[INFO] Static IP address has been configured, will use 192.168.178.49
[INFO] Trying to connect to WiFi with SSID 'REAVER' .....................................................
[WARNING] Connection to WiFi network with SSID 'REAVER' failed with timeout
[WARNING] starting server in access point mode with SSID 'ESPStepperMotorServer' and password 'Aa123456' as fallback
[INFO] Started Access Point with name ESPStepperMotorServer and IP 192.168.4.1
[INFO] Listing files in root folder of SPIFFS:
[INFO] Starting webserver on port 80
[INFO] The file /index.html could not be found on SPIFFS
[INFO] The file /js/app.js.gz could not be found on SPIFFS
[INFO] The file /img/logo.svg could not be found on SPIFFS
[INFO] The file /favicon.ico could not be found on SPIFFS
[INFO] The file /img/rotaryEncoderWheel.svg could not be found on SPIFFS
[INFO] The file /img/emergencyStopSwitch.svg could not be found on SPIFFS
[INFO] The file /img/stepper.svg could not be found on SPIFFS
[INFO] The file /img/switch.svg could not be found on SPIFFS
[INFO] The file /upload.html.gz could not be found on SPIFFS
[WARNING] The UI does not seem to be installed completely on SPIFFS. Automatic download failed since the server is in Access Point mode and not connected to the internet
[WARNING] Start the server in wifi client (STA) mode to enable automatic download of the web interface files to SPIFFS
[INFO] No web UI could be registered
[INFO] Webserver started, you can now open the user interface on http://192.168.4.1:80/
[INFO] Command Line Interface started, registered 22 commands. Type 'help' to get a list of all supported commands
[INFO] Motion Controller task started
................................................
[WARNING] Connection to WiFi network with SSID 'REAVER' failed with timeout
[WARNING] starting server in access point mode with SSID 'ESPStepperMotorServer' and password 'Aa123456' as fallback
[INFO] Started Access Point with name ESPStepperMotorServer and IP 192.168.4.1
[INFO] Listing files in root folder of SPIFFS:
[INFO] Starting webserver on port 80
[INFO] The file /index.html could not be found on SPIFFS
[INFO] The file /js/app.js.gz could not be found on SPIFFS
[INFO] The file /img/logo.svg could not be found on SPIFFS
[INFO] The file /favicon.ico could not be found on SPIFFS
[INFO] The file /img/rotaryEncoderWheel.svg could not be found on SPIFFS
[INFO] The file /img/emergencyStopSwitch.svg could not be found on SPIFFS
[INFO] The file /img/stepper.svg could not be found on SPIFFS
[INFO] The file /img/switch.svg could not be found on SPIFFS
[INFO] The file /upload.html.gz could not be found on SPIFFS
[WARNING] The UI does not seem to be installed completely on SPIFFS. Automatic download failed since the server is in Access Point mode and not connected to the internet
[WARNING] Start the server in wifi client (STA) mode to enable automatic download of the web interface files to SPIFFS
[INFO] No web UI could be registered
[INFO] Webserver started, you can now open the user interface on http://192.168.4.1:80/
[INFO] Command Line Interface started, registered 22 commands. Type 'help' to get a list of all supported commands
[INFO] Motion Controller task started
E (144) psram: PSRAM ID read error: 0xffffffff
[INFO] Setting log level to INFO
[WARNING] Failed to load configuration file from SPIFFS. File /config.json not found or SPIFFS is not enabled/mounted
[INFO] Starting ESP-StepperMotor-Server (v. 0.4.6)
[INFO] Static IP address has been configured, will use 192.168.1.10
[INFO] Trying to connect to WiFi with SSID 'REAVER' ...........E (144) psram: PSRAM ID read error: 0xffffffff
[INFO] Setting log level to INFO
[WARNING] Failed to load configuration file from SPIFFS. File /config.json not found or SPIFFS is not enabled/mounted
[INFO] Starting ESP-StepperMotor-Server (v. 0.4.6)
[INFO] Static IP address has been configured, will use 192.168.1.10
[INFO] Trying to connect to WiFi with SSID 'Reaver' .............
[INFO] Connected to network with IP address 192.168.1.10
[INFO] Listing files in root folder of SPIFFS:
[INFO] Starting webserver on port 80
[INFO] The file /index.html could not be found on SPIFFS
[INFO] Download of /index.html completed
[INFO] The file /js/app.js.gz could not be found on SPIFFS
[INFO] Download of /js/app.js.gz completed
[INFO] The file /img/logo.svg could not be found on SPIFFS
[INFO] Download of /img/logo.svg completed
[INFO] The file /favicon.ico could not be found on SPIFFS
[INFO] Download of /favicon.ico completed
[INFO] The file /img/rotaryEncoderWheel.svg could not be found on SPIFFS
[INFO] Download of /img/rotaryEncoderWheel.svg completed
[INFO] The file /img/emergencyStopSwitch.svg could not be found on SPIFFS
[INFO] Download of /img/emergencyStopSwitch.svg completed
[INFO] The file /img/stepper.svg could not be found on SPIFFS
[INFO] Download of /img/stepper.svg completed
[INFO] The file /img/switch.svg could not be found on SPIFFS
[INFO] Download of /img/switch.svg completed
[INFO] The file /upload.html.gz could not be found on SPIFFS
[INFO] Download of /upload.html.gz completed
[INFO] Webserver started, you can now open the user interface on http://192.168.1.10:80/
[INFO] Command Line Interface started, registered 22 commands. Type 'help' to get a list of all supported commands
[INFO] Motion Controller task started
E (144) psram: PSRAM ID read error: 0xffffffff
[INFO] Setting log level to INFO
[WARNING] Failed to load configuration file from SPIFFS. File /config.json not found or SPIFFS is not enabled/mounted
[INFO] Starting ESP-StepperMotor-Server (v. 0.4.7)
[INFO] Static IP address has been configured, will use 192.168.1.10
[INFO] Trying to connect to WiFi with SSID 'Reaver' ............
[INFO] Connected to network with IP address 192.168.1.10
[INFO] Listing files in root folder of SPIFFS:
[INFO] File: /index.html (0) -1
[INFO] File: /js/app.js.gz (0) -1
[INFO] File: /img/logo.svg (0) -1
[INFO] File: /favicon.ico (0) -1
[INFO] File: /img/rotaryEncoderWheel.svg (0) -1
[INFO] File: /img/emergencyStopSwitch.svg (0) -1
[INFO] File: /img/stepper.svg (0) -1
[INFO] File: /img/switch.svg (0) -1
[INFO] File: /upload.html.gz (0) -1
[INFO] Starting webserver on port 80
[INFO] Webserver started, you can now open the user interface on http://192.168.1.10:80/
[INFO] Command Line Interface started, registered 22 commands. Type 'help' to get a list of all supported commands
[INFO] Motion Controller task started
E (144) psram: PSRAM ID read error: 0xffffffff
[INFO] Setting log level to INFO
[WARNING] Failed to load configuration file from SPIFFS. File /config.json not found or SPIFFS is not enabled/mounted
[INFO] Starting ESP-StepperMotor-Server (v. 0.4.7)
[INFO] Trying to connect to WiFi with SSID 'Reaver' ..................
[INFO] Connected to network with IP address 192.168.1.93
[INFO] Listing files in root folder of SPIFFS:
[INFO] File: /index.html (0) -1
[INFO] File: /js/app.js.gz (0) -1
[INFO] File: /img/logo.svg (0) -1
[INFO] File: /favicon.ico (0) -1
[INFO] File: /img/rotaryEncoderWheel.svg (0) -1
[INFO] File: /img/emergencyStopSwitch.svg (0) -1
[INFO] File: /img/stepper.svg (0) -1
[INFO] File: /img/switch.svg (0) -1
[INFO] File: /upload.html.gz (0) -1
[INFO] Starting webserver on port 80
[INFO] Webserver started, you can now open the user interface on http://192.168.1.93:80/
[INFO] Command Line Interface started, registered 22 commands. Type 'help' to get a list of all supported commands
[INFO] Motion Controller task started
```

pkerspe, 2021-12-18:
You have some PSRAM error showing, I suggest you solve that first, this is not related to the library but seems some general problem in you IDE setup / programmer setup. Maybe you configured PSRAM but your device does not have any?
Also you can manually flash the UI files in spiffs since in the beginning it seems you did not do do that and the logs indicated that the UI is missing. After attempt from the library to download the files manually the size shows as 0/-1 which seems not correct.
Also upload a config Jain file to spiffs to get rid of that error message

FigJam23, 2021-12-18:
> You have some PSRAM error showing, I suggest you solve that first, this is not related to the library but seems some general problem in you IDE setup / programmer setup. Maybe you configured PSRAM [...]

hey, thanks

after flashing to a different esp32 all is working now....
any idea how I could add controls via keyboard or a joystick if even possible thanks

pkerspe, 2021-12-19:
Well, joystick or keyboard support is doable depending on the type of interface you want to use.
If you use an old "non usb" joystick (15 pin connectors) you might be able to arrange for that with the regular IO Pins of the esp32, or use a ps2 keyboard and check for an PS2 library for the ESP. Adding USB-host support for HID USB to an esp32 might be tricky and requires additional external hardware most likely. Never tried that. 
The "new" ESP-S2 has native USB support, you might get lucky with that one as well.
Probably the best way to go would be to go with a BLE keyboard and use for example this library as a first start: https://github.com/chegewara/esp32-hid-keyboard-client
But you are in your own there, I do not have practical experience with these type of interface devices and the esp32

FigJam23, 2021-12-19:
> Well, joystick or keyboard support is doable depending on the type of interface you want to use. If you use an old "non usb" joystick (15 pin connectors) you might be able to arrange for that with the regular IO Pins of the esp32, or use a ps2 keyboard and check for an PS2 library for the ESP (e.g. https://github.com/michalhol/ps2kbdlib ), there are also cheap modules to use ps2 keyboards via I2C, search for "PS2 Keyboard Driver Module Serial Port" on eBay (cost 3-4 bucks). Adding USB-host support for HID USB to an esp32 might be tricky and requires additional external hardware most likely. Never tried that. The "new" ESP-S2 has native USB support, you might get lucky with that one as well. Probably the best way to go would be to go with a BLE keyboard and use for example this library as a first start: https://github.com/chegewara/esp32-hid-keyboard-client But you are in your own there, I do not have practical experience with these type of interface devices and the esp32 and all links provided above are untested, just first hits that showed up on Google search.

thanks mate much appreciate the info ill have a play see what happens

closed 2021-12-19 by pkerspe