pkerspe/ESP-StepperMotor-Server #33: Possibility to control enable pin on the A4988 driver module. Otherwise the motor uses power when not rotating.

amigafan, 2022-05-26:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

pkerspe, 2022-05-27:
@amigafan can you please provide some details on your feature request?

amigafan, 2022-05-27:
My idea is to use an A4988 driver module together with a DRV8825 expansion board module.
This expandion module requires Step signal, Direction signal and an enabled signal. When enable signal is low the driver is enabled. If the enable signal is low all the time, the step motor uses power even when it is not rotating. I like the ability to configure stepper motor, limit switch and rotary encoder. Br. amigafan

amigafan, 2022-05-27:
Yes, that is correctly understood. For some reason, the motor draws power as soon as the enable pin is low.

pkerspe, 2022-05-28:
Probably the driver powers the coils to keep the motor from spinning freely, in order to maintain its position. If you remove power, depending on your setup and use case, you might loose absolute precision.

amigafan, 2022-05-28:
That sounds like a good explanation, without power the shaft is easy to turn around with your fingers. For my use, though, it's no problem.

pkerspe, 2023-06-30:
so for everyone having this issue you can now use the motor break function to control a driver enable pin. Basically instead of using it to activate a break you use it to control the state of the enable pin when the motor is not moving:
  `void setBrakePin(signed char brakePin, byte activeState = ESP_FlexyStepper::ACTIVE_HIGH);`

closed 2023-06-30 by pkerspe