pkerspe/ESP-StepperMotor-Server #36: Adding a Joystick for XYZ-control

maxholgasson, 2023-01-03:
Hi,
I would like to control 3 axis with a joystick and therefore need 3 analog input pins. Is it possible to add that?

pkerspe, 2023-01-03:
Can you explain a bit more in detail what the analog inputs relation to the stepper motors movement is supposed to look like?
Would it control rotation speed and direction only? Or rather move to a specific position according to the analog input and stay there as long as the analog value does not change?
Also which configuration parameters would you need for setting up the link between analog input and a stepper motor? (E.g. define analog range and direction / speed of stepper etc.)
I would say if the analog inputs "only" define the direction and speed of rotation it would be simpler to implement.

maxholgasson, 2023-01-03:
I would have an use case for your first suggestion. So it would control speed and direction.
I'm using this joysticks https://www.aliexpress.com/item/4000825085381.html
I did that with an 328P but with that microcontroller the max output frequency is limited especially if you're moving 3 axis simultaneously. I guess an ESP32 would have more power.
As configuration parameters I could image as you said speed of stepper/steps needed per revolution and what would be amazing is a kind of ramp for starting and stopping.I'd like to mount a camera on the axis and be able to move it very smooth with a soft start and stop

pkerspe, 2023-01-03:
The smooth start and stop might add a bit to the challenge. 
In general i think you could basically implement all this in code with the existing esp flexy stepper library.
Basically just read the analog signals in the main loop and set the speed and direction according to your own choice of mapping table (mapping analog value to a speed and direction) and update the speed parameter in the flexy stepper while increasing/decreasing the target position value as long as the joystick is not in a neutral position (the you set the target position to the current position).
The acceleration / deceleration settings from the esp flexystepper library will make sure the start and stop are smooth.

pkerspe, 2023-01-03:
Please note: when you want realy smooth movements you should not use wifi / webserver etc running on the ESP, so using the flexy stepper library directly would be better rather than using the stepper motor sever.
See also the issue about jitter for more details on clean step signal generation

maxholgasson, 2023-01-04:
Ok thanks for the explanation. Also the advice with your flexystepper. I liked your web ui very much and thought it could be handy maybe as a second option to be able to control it in certain circumstances also via web. I got your concerns with the jitter. That seems a big bumper and I guess it wouldnt work reliably with an ESP then. Guess switching to an SAMD21 or so would make more sense

closed 2023-06-30 by pkerspe