pkerspe/ESP-StepperMotor-Server #37: Can't build on PlatformIO

tiberiust, 2023-01-24:
guten Tag. 
This looks just what I need to put a bunch of steppers thru their basic paces. But build always fails with numerous "no type named 'type' in 'struct ArduinoJson..." errors

`[{
	"resource": "/home/markt/Synced/VSCode/PlatformIO/Office/Stepper/.pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson/Variant/VariantRefBase.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson6200_F1::enable_if<false, char*>'",
	"startLineNumber": 53,
	"startColumn": 3,
	"endLineNumber": 53,
	"endColumn": 3
},{
	"resource": "/home/markt/Synced/VSCode/PlatformIO/Office/Stepper/.pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson/Variant/VariantRefBase.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson6200_F1::enable_if<false, char*>'",
	"startLineNumber": 61,
	"startColumn": 3,
	"endLineNumber": 61,
	"endColumn": 3
},{
	"resource": "/home/markt/Synced/VSCode/PlatformIO/Office/Stepper/.pio/libdeps/esp32dev/ESP-StepperMotor-Server/src/ESPStepperMotorServer_Configuration.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no matching function for call to 'ArduinoJson6200_F1::JsonVariant::as<char*>()'",
	"startLineNumber": 263,
	"startColumn": 51,
	"endLineNumber": 263,
	"endColumn": 51
}`
etc etc ... There's about another 10 errors. All with message 'no matching function call ArduinoJson...'

**Environment**
Suse Linux. Kernel 6.1. VSCode, PlatformIO 6.1.5. All defaults. 

**To Reproduce**
To test if my Linux setup was causing problems, I setup a completely fresh VSCode + PlatformIO on a Win10 box. Result was the same.

**Expected behavior**
Compile ;)

**PlatformIO.ini:**
```
[env:esp32dev]
platform = espressif32
board = esp32dev 
framework = arduino
lib_deps =  https://github.com/pkerspe/ESP-StepperMotor-Server.git, bblanchon/ArduinoJson@^6.20.0 #tried with and without specifying the JSON lib. 
monitor_speed = 115200
```
I also tried other boards. No difference.

**main.cpp:**
```
#include <ESPStepperMotorServer.h>
//#include <ArduinoJson.h> //tried with and without

ESPStepperMotorServer *stepperMotorServer;

const char *wifiName= "mySSID"; // enter the SSID of the wifi network to connect to
const char *wifiSecret = "mypassword"; // enter the password of the the existing wifi network here

void setup() 
{
  Serial.begin(115200);
  stepperMotorServer = new ESPStepperMotorServer(ESPServerRestApiEnabled | ESPServerWebserverEnabled | ESPServerSerialEnabled);
  stepperMotorServer->setWifiCredentials(wifiName, wifiSecret);
  stepperMotorServer->setWifiMode(ESPServerWifiModeClient); //start the server as a wifi client (DHCP client of an existing wifi network)
  stepperMotorServer->start();
}

void loop() 
{
}
````

**Everything seems to be in the right place:**
`markt@mtdesk:~/Synced/VSCode/PlatformIO/Office/Stepper/.pio/libdeps/esp32dev> ls
ArduinoJson  AsyncTCP  ESP Async WebServer  ESP-FlexyStepper  ESP-StepperMotor-Server  integrity.dat`

I have used this Linux setup to build and upload several simple ESP and Arduino programs without problems in the past and it still works fine. But ... I am just a few pages on from beginner at cpp and PlatformIO so I might be doing something newb/dumb. Googling the compile error seems to point to a coding error [somewhere](https://stackoverflow.com/questions/56117241/no-type-named-type-in-struct-stdenable-iffalse-void) 

Thanks in advance anyone.

pkerspe, 2023-01-24:
Hi @tiberiust,
the cause of the error lies within the ArduinoJSON library as far as I can see and not in the ESP Stepper motor server code.
I suggest you open a ticket in the repository of ArduinoJSON.
Also you can try to create basic project only using ArduinoJSON library and not the ESP Stepper Motor Server library, to isolate the issue and see if there is any relation to the ESP Stepper Motor Server.

pkerspe, 2023-01-24:
on second view you might also want to try using an older version of Arduino JSON. Locally I use ArduinoJson @ 6.16.1 and that seems to work. 
I will need to test a bit more if newer versions have breaking changes.

tiberiust, 2023-01-24:
Indeed. I successfully built it on ArduinoIDE with 6.19.4. On PlatformIO I didn't have much luck (lots of Async Webserver errors) until I found this [https://github.com/me-no-dev/ESPAsyncWebServer/issues/1151](url) and changed the platformIO.ini file to

```
[env:esp32-s3-devkitc-1] 
platform = espressif32 
board = esp32-s3-devkitc-1
lib_deps = 
    ESP-StepperMotor-Server ;
    ArduinoJson@6.16.1 ;
    https://github.com/me-no-dev/ESPAsyncWebServer.git ;
framework = arduino
monitor_speed = 115200
```

But that still didn't give me a working webserver. Only when I set a static IP as per your example did it all work. 

I don't have the experience to know what's happening, but it's working and I'll start playing.

Thanks a lot pkerspe :)

pkerspe, 2023-06-30:
the error should be fixed in latest version 0.4.11
I updated dependencies to version ArduinoJSON 6.21.2 and fixed the deprecation warnings.

**Please also look at release notes for infos on potential errors in relation to some missing SAH1 declarations in ESPAsyncWebserver!**

closed 2023-06-30 by pkerspe