pkerspe/ESP-StepperMotor-Server #38: Program additional functions

xpix, 2023-02-03:
Hello Paul,

I would like to program an additional function (solar tracker), how would you go about it. So I guess I need a separate module, but at which point should I put it in? Besides of course extend the UI, but that I should deal with the UI.

If you would make me there a short suggestion, I would be very grateful :)

Greetings
Frank

pkerspe, 2023-02-03:
Hi Frank,
a solar tracker seems to be a rather specific function and I am not sure if it should be a flow blown feature in the stepper motor server.
I'd rather call it "automatic position adjustment".
So we could add support for an analog value that can be linked to a Stepper Axis and basically moves the stepper as long as the analog value is above of below a certain threshold.

How would you implement the solar tracker? Simple LDR (Photoresistiors) Setup or something more sophisticated?
As for the UI what kind of functions would you like to add to the UI?

xpix, 2023-02-05:
Hello paul,
so I just forked your project and want to add an additional feature in my
version, regardless of your version. Just4Fun :)

So, I have developed a stepper motor or rather servo (Xmoto), which
consists of a gear DC motor with extended shaft and magnetic disk as well
as motor driver and MCu. Here then the revolutions of the magnetic disk are
translated into steps and also the position is held. Here the Hackaday page
to it:
https://hackaday.com/2022/01/01/teaching-a-dc-servo-motor-to-act-like-a-stepper/

I would start from the manually entered GPS position and get the time via
ntp, then the azimuth and elevation angle of the sun is calculated and the
motors are set that way.

In the UI I would like to have the GPS data as well as Altitude entered and
saved. An extra page in the UI would then display the calculated sun
position and allow reference runs. Something like this:
https://github.com/happytm/SunTracker

My question, how would you best implement this in your project, where would
you put the SunCalc logic? I already tried to read out the GPS position
seriel in the loop, but that seemed to influence your program too much.

I look forward to your answer
Frank

*--*
*Frank Herrmann    | Software Entwicklung*
+49 151 649 36606 | ***@***.***
Neudorfer Weg.10  | 68753 Waghäusel

pkerspe, 2023-02-06:
Hi Frank,

When you talk a lot reading gps via serial, are you referring a gps receiver connected via serial interface or do you mean you wanted to input some coordinates via the serial interface?

To create a new view in the webinterface you can just add a new vue.js view with the needed form elements, then rebuild the vue application and upload to the esp32.
In order to communicate with the esp32 stepper server you need to add some rest endpoints to handle the requests from the frontend in the function:

ESPStepperMotorServer_RestAPI::registerRestEndpoints()

xpix, 2023-02-07:
Hi Paul,

Well, I have tried to read out the data seriel via a GPS sensor. But since my solar tracker is mostly stationary, it is also sufficient to simply enter the data manually in the UI and save it. Since we have a client/server system, it is quite easy to extend the UI and control the ESP via REST interface. 

Thank you very much for your help.

Greetings
Frank