pkerspe/ESP-StepperMotor-Server #40: hi i coudnt load to esp nodemcu-32s

kucukonur, 2023-11-20:
hi  how are you doing ? 
i liked ur project 
i wanted to try to your  esp stepper server  but i coudnt load i tried so much thing  my  esp = nodemcu-32s 

But first, I need to upload your code into the NodeMCU-32S. I'm encountering the following error. Could you please help?"

: C:/Users/user/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:48: undefined reference to `loop()'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32devServer\firmware.elf] Error 1

can u load a video from youtube how can we  load ? 

than can u help me for this project after project i can give donate 100 €

"I want to drive two stepper motors simultaneously. For motor 1, I will input speed and direction information through the interface, and I want to read the angle of rotation from the encoder. I need to modify the web interface as well. For motor 2, I will connect it to a linear axis, and it will move 13.08cm with each step. The linear axis is 2 meters long. Motor 2 will move back and forth between two positions entered in the interface at the desired speed on the linear axis. How can I make these changes?
you can contact of me from mail to  onurkucuk58@gmail.com 
im working in vocational high school :))

pkerspe, 2023-11-20:
Please post your code for main.cpp so I can see what is going on.
Seems you are trying to call loop() function before it is defined/declared

kucukonur, 2023-11-20:
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_task_wdt.h"
#include "Arduino.h"
#if (ARDUINO_USB_CDC_ON_BOOT|ARDUINO_USB_MSC_ON_BOOT|ARDUINO_USB_DFU_ON_BOOT)
&& !ARDUINO_USB_MODE
#include "USB.h"
#if ARDUINO_USB_MSC_ON_BOOT
#include "FirmwareMSC.h"
#endif
#endif

#ifndef ARDUINO_LOOP_STACK_SIZE
#ifndef CONFIG_ARDUINO_LOOP_STACK_SIZE
#define ARDUINO_LOOP_STACK_SIZE 8192
#else
#define ARDUINO_LOOP_STACK_SIZE CONFIG_ARDUINO_LOOP_STACK_SIZE
#endif
#endif

TaskHandle_t loopTaskHandle = NULL;

#if CONFIG_AUTOSTART_ARDUINO
#if CONFIG_FREERTOS_UNICORE
void yieldIfNecessary(void){
    static uint64_t lastYield = 0;
    uint64_t now = millis();
    if((now - lastYield) > 2000) {
        lastYield = now;
        vTaskDelay(5); //delay 1 RTOS tick
    }
}
#endif

bool loopTaskWDTEnabled;

__attribute__((weak)) size_t getArduinoLoopTaskStackSize(void) {
    return ARDUINO_LOOP_STACK_SIZE;
}

void loopTask(void *pvParameters)
{
    setup();
    for(;;) {
#if CONFIG_FREERTOS_UNICORE
        yieldIfNecessary();
#endif
        if(loopTaskWDTEnabled){
            esp_task_wdt_reset();
        }
        loop();
        if (serialEventRun) serialEventRun();
    }
}

extern "C" void app_main()
{
#if ARDUINO_USB_CDC_ON_BOOT && !ARDUINO_USB_MODE
    Serial.begin();
#endif
#if ARDUINO_USB_MSC_ON_BOOT && !ARDUINO_USB_MODE
    MSC_Update.begin();
#endif
#if ARDUINO_USB_DFU_ON_BOOT && !ARDUINO_USB_MODE
    USB.enableDFU();
#endif
#if ARDUINO_USB_ON_BOOT && !ARDUINO_USB_MODE
    USB.begin();
#endif
    loopTaskWDTEnabled = false;
    initArduino();
    xTaskCreateUniversal(loopTask, "loopTask", getArduinoLoopTaskStackSize(),
NULL, 1, &loopTaskHandle, ARDUINO_RUNNING_CORE);
}

#endif

Paul ***@***.***>, 20 Kas 2023 Pzt, 21:45 tarihinde şunu
yazdı:

kucukonur, 2023-11-20:
i downloaded [ESP_StepperMotor_Server-0.4.11.zip](https://downloads.arduino.cc/libraries/github.com/pkerspe/ESP_StepperMotor_Server-0.4.11.zip)  

i didnt change nothing i coudnt loaded

pkerspe, 2023-11-20:
- You are using freertos logic directly but it seems you are not so much aware of what you are doing. Seems you did not define any loop function as far as I can see.
Please start of with one of the examples provided in the library, just rename to main.cpp and use it on platform io then try to build.

kucukonur, 2023-11-20:
"I downloaded the code without making any changes. I am trying to upload it without making any modifications, but I continue to encounter errors

what can i do i dont know 
i m trying to load since a day

pkerspe, 2023-11-20:
You don't need to download any code, platformio does dependency management for you if you follow the instructions from the readme and simply use the dependency management in platformio.ini file.
Follow readme step by step as described in the firmware upload section

kucukonur, 2023-11-20:
"Can I directly upload the downloaded zip file from PlatformIO? The
instructions are too long, so I wanted to upload the project directly. Can
you add the variables to the interface and modify the project for a fee? My
time is limited."

Paul ***@***.***>, 20 Kas 2023 Pzt, 23:24 tarihinde şunu
yazdı:

pkerspe, 2023-11-20:
The zip file you downloaded is not a project file, it is just a library. You use the library in you own project. The manual for setting up the project is basically 4-5 steps, you just need to read the manual, would take less than 5 minutes.
I am afraid I cannot implement your requested project for the given fee, you project seems much more complex than just using the library as-is. Maybe you can post your job on platforms like upload to find a developer with a lower hourly fee that can implement it for the given budget.
I will close this issue now, since it is not an issue of the library itself.

closed 2023-11-20 by pkerspe