pkerspe/ESP-StepperMotor-Server #45: Can't install in arduino

iplayfast, 2024-01-25:
**Describe the bug**
try to install into arduino ide, 
The ide sees the library, but when I click install I get error message 
"Failed to install library: 'ESP-StepperMotor-Server:0.4.12'. No valid dependencies solution found: dependency 'ArduinoJSON' is not available"

**Environment**
Arduino

pkerspe, 2024-01-25:
Did you install ArduinoJSON before?

vgegok, 2024-01-27:
the same problem：
[image]

Zanacks1, 2024-08-19:
Same, tried several versions of ArduinoJson 6.x, 7.x, couple of betas.. No Joy..

rafik24, 2024-11-04:
Same here, tried on arduino IDE and vscode/platformio and I can say there is no way to build a running code.

The best I had is arduinojson throwing a lot of warnings which I interpret more as errors and the esp32 boards crashing on startup.

I used esp-idf 5.3.2 if it can help but after several trials, library versions and all as per the install instructions nothing works unfortunately.

pkerspe, 2024-11-04:
Please provide details on the errors you get in VSC/ PlatformIO

rafik24, 2024-11-05:
> Please provide details on the errors you get in VSC/ PlatformIO

Hi pkerspe,

Here is the serial output from vscode:

--- Terminal on COM11 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
[INFO] Setting log level to INFO
[INFO] Loading configuration file /config.json from SPIFFS
[INFO] Serializing config
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008a0f8  PS      : 0x00060830  A0      : 0x800e35f9  A1      : 0x3ffb1fa0  
A2      : 0x00000000  A3      : 0x3ffb2438  A4      : 0x00000004  A5      : 0x3ffb1f90  
A6      : 0x00000004  A7      : 0x929776ab  A8      : 0x00000001  A9      : 0x3ffb1b50  
A10     : 0x3f4153a2  A11     : 0x3ffb1b84  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x3ffc3c54  A15     : 0x00000004  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4008a0f8  LEND    : 0x4008a106  LCOUNT  : 0x00000000

Backtrace: 0x4008a0f5:0x3ffb1fa0 0x400e35f6:0x3ffb1fc0 0x400d8ef1:0x3ffb1fe0 0x400da8f5:0x3ffb2080 0x400db4b3:0x3ffb21e0 0x400d3be6:0x3ffb2220 0x400d31ae:0x3ffb2240 0x400e6db7:0x3ffb2270 0x4009085e:0x3ffb2290
  #0  0x4008a0f5 in memcmp at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/string/memcmp.c:73
  #1  0x400e35f6 in IPAddress::operator==(unsigned char const*) const at C:/Users/User/.platformio/packages/framework-arduinoespressif32/cores/esp32/IPAddress.cpp:275 (discriminator 
1)
  #2  0x400d8ef1 in ESPStepperMotorServer_Configuration::serializeServerConfiguration(ArduinoJson::V6212PB::JsonDocument&, bool) at .pio/libdeps/wemos_d1_mini32/ESP-StepperMotor-Server/src/ESPStepperMotorServer_Configuration.cpp:101
  #3  0x400da8f5 in ESPStepperMotorServer_Configuration::loadConfiguationFromSpiffs(String) at .pio/libdeps/wemos_d1_mini32/ESP-StepperMotor-Server/src/ESPStepperMotorServer_Configuration.cpp:245
  #4  0x400db4b3 in ESPStepperMotorServer_Configuration::ESPStepperMotorServer_Configuration(char const*, bool) at .pio/libdeps/wemos_d1_mini32/ESP-StepperMotor-Server/src/ESPStepperMotorServer_Configuration.cpp:45
  #5  0x400d3be6 in ESPStepperMotorServer::ESPStepperMotorServer(unsigned char, unsigned char) at .pio/libdeps/wemos_d1_mini32/ESP-StepperMotor-Server/src/ESPStepperMotorServer.cpp:131
  #6  0x400d31ae in setup() at src/main.cpp:37
  #7  0x400e6db7 in loopTask(void*) at C:/Users/User/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:59
  #8  0x4009085e in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162  
  
  I get a lots of warnings from arduinojson, the code compiles but I get this mediation guru error on several kind of ESP32 boards.
  
Thank you,
Rafik

pkerspe, 2024-11-05:
Those are not warnings but a kernel panic resulting in the esp32 to reboot. 
The panic seems to occur while accessing the SPIFFFS to load a config.
Bit there is also an error in some ipaddress class.
Two things:
Please post your main.cpp file 
Please verify you actually have formatted your spiffs and uploaded a config file in there

rafik24, 2024-11-05:
> Those are not warnings but a kernel panic resulting in the esp32 to reboot. The panic seems to occur while accessing the SPIFFFS to load a config. Bit there is also an error in some ipaddress class. [...]

I wanted to report thank deleting platformio and ESP-IDF extensions in VSCode, then uninstalling vscode and deleted all remaining folders I managed to get up and running.

You were right, it seem like the SPIFFS image that was downloaded had an issue somehow and nothing would help other than deleting all and re-installing.

While vscode/platformio make things simpler they are really a pain to troubleshoot when something goes wrong :-)

Many thanks for your support pkerspe and the amazing project you built !

All the Best,
Rafik