pkerspe/ESP-StepperMotor-Server #46: Port over to FastAccelStepper

dronus, 2024-07-10:
Currently, the used stepper library is CPU-bound and does not cooperate well with other CPU tasks. This results in stepper pulse jitter even on quite low stepping frequencies, making microstepping quite unusable, when for example using the web interface, which puts load on CPU. 

On contrast, FastAccelStepper uses ESP hardware units to generate the pulse train, and is quite immune to short CPU load.  Especially, using AsyncWebserver over WiFi does not add much jitter, even at 50000 pulses / second (numbers by own experimentation), which is much faster that currently possible with FlexyStepper. This makes microstepping feasable, resulting in much smoother motor operation.

As FastAccelStepper does not provide unit conversion as far as I understand, those would have to be done in ESP-StepperMotor-Server code.

pkerspe, 2024-07-10:
Thank you for your input, I am absolutely willing to consider an integration with another stepper library! Please create a pull request for a first POC and we can review jointly.

Anasrais-hub, 2024-10-16:
I have been trying to integrate the accelstepper library along with other changes but as I am not a programmer I have been struggling a lot. I am willing to finance a programmer to do it, so if one of you are interested or know someone who can do it I would be glad to work it out.

dronus, 2025-02-15:
Hi! Sorry for the late answer. 

I've switched over to developing my own wifi motor controller, which is currently work-in-progress here: https://github.com/dronus/jammotors - but that changed scope in favor of supporting other motor driver types and protocols, and not establishing a sleek UI so far. It does not convert units. 

So it would still be reasonable for me to port [ESP-StepperMotor-Server](https://github.com/pkerspe/ESP-StepperMotor-Server) over to FastAccelSteppers. The main culprit is the support for physical units, which is of course a good feature.