pkerspe/ESP-StepperMotor-Server #47: Support for magnetic encoders over i2c

rafik24, 2024-11-05:
**Is your feature request related to a problem? Please describe.**
Hi, are there any plans to support for magnetic encoders over i2c like the AS500

**Describe the solution you'd like**
Ability to add a magnetic encoder and define the i2c bus and address it can be reached on

Thanks,
Rafik

pkerspe, 2024-11-05:
What exactly is the purpose of adding the encoder? What use case do you want cover with the encode?
At the moment it is not planned but it you have a valid use case we can discuss efforts

rafik24, 2024-11-05:
Hi Paul,

The idea behind it would be that the motor could drive and be driven while reporting the position/angle and velocity.

Thanks,
Rafik

pkerspe, 2026-02-13:
Closing this for now since no further feedback in requirement details was provided.

closed 2026-02-13 by pkerspe